dotintent/react-native-ble-plx #90: Execution failed for task ':react-native-ble-plx:compileReleaseAidl'.

inev, 2017-06-17:
When I try to add the library to an existing `ignite` project and try to run via `react-native run-android` I get `Execution failed for task ':react-native-ble-plx:compileReleaseAidl'.`

I looked around and the common suggestion is to upgrade the build tools version but react native wants that fixed.  Any suggestions?

mhyousefi, 2017-07-17:
I have the same issue!
* What went wrong:
Execution failed for task ':react-native-icons:compileReleaseAidl'.
> java.lang.IllegalStateException: aidl is missing

Cierpliwy, 2018-04-09:
Please retest with latest version and reopen if needed.

closed 2018-04-09 by Cierpliwy

PradeepNedun, 2018-06-22:
@Cierpliwy  I am still getting the same error with the latest version too. Please re-open this if not fixed.