dotintent/react-native-ble-plx #91: Received data is incomplete (monitorCharacteristicForDevice)

hhtu, 2017-06-17:
Hi 

I use monitorCharacteristicForDevice to receive my car dongle notifications, and there are something wrong when I try to query for DTC(diagnosis trouble code)

I have a simulator which will send 7 error set

When I send AT command: 30 33 (MDMN) (using writeCharacteristicWithoutResponseForDevice)

It should be return:

03\r02A \r
0: 43 14 00 43 00 92 \r
1: 10 43 20 43 30 43 40 \r
2: 43 40 92 50 43 60 43 \r
3: 70 43 80 43 80 92 90 \r
4: 43 A0 43 B0 43 C0 43 \r
5: C0 92 D0 43 E0 43 F0 \r
6: 43 99 99 99 99 99 99 \r\r>

but the weird thing is - Sometimes, monitorCharacteristicForDevice give me different results

ex:

03\r02A \r
0: 43 14 00 43 00 92 \r
1: 10 43 20 43 30 43 40 \r
4: 43 A0 43 B0 4\r>

or

03\r02A \r
0: 43 14 00 43 00 92 \r
1: 10 43 20 43 30 43 40 \r
2: \r>

or ...etc

I have to try several times to get correct answer, can you help me how to solve this problem?
(The weird thing is...somehow I can get correct answer)

(I use Bluetooth HCI sniffer to check correct answer)

I use base-64 library to combine characteristic.value (base64.decode(characteristic.value))

Cierpliwy, 2017-06-17:
First of all I would try to enable native module's logging mechanism:
`yourManager.setLogLevel(LogLevel.Verbose)`
and search for logs which looks like: 
`Notification from Characteristic(uuid: xxxx, id: xxxx, value: your hex value);`

There will be plenty of other messages related to BLE stack. Try to send me as much information as you can by doing that.

hhtu, 2017-06-17:
Hi 

It seems this one could be the clue: 06-17 21:25:25.334  6651  6792 E bt_att  : invalid response/indicate pkt size: 53, PDU size: 23

Here is one example

06-17 21:25:25.243 31817  1997 V RxBle#BleModule: Notification from Characteristic(uuid: 00002af0-0000-1000-8000-00805f9b34fb, id: 21, value: 30330D)
06-17 21:25:25.245 31817 31889 D RxBle#BluetoothGatt: onCharacteristicChanged characteristic=00002af0-0000-1000-8000-00805f9b34fb
06-17 21:25:25.247 31817 31889 D RxBle#BluetoothGatt: onCharacteristicChanged characteristic=00002af0-0000-1000-8000-00805f9b34fb
06-17 21:25:25.247 31817  1997 V RxBle#BleModule: Notification from Characteristic(uuid: 00002af0-0000-1000-8000-00805f9b34fb, id: 21, value: 303241200D303A20)
06-17 21:25:25.250 31817  1997 V RxBle#BleModule: Notification from Characteristic(uuid: 00002af0-0000-1000-8000-00805f9b34fb, id: 21, value: 3433203134203030)
06-17 21:25:25.258  6651  6792 E bt_btif : bta_gattc_process_indicate, ignore HID ind/notificiation
06-17 21:25:25.259  6651  6792 E bt_btif : bta_gattc_process_indicate, ignore HID ind/notificiation
06-17 21:25:25.260 31817 31832 D RxBle#BluetoothGatt: onCharacteristicChanged characteristic=00002af0-0000-1000-8000-00805f9b34fb
06-17 21:25:25.261 31817  1997 V RxBle#BleModule: Notification from Characteristic(uuid: 00002af0-0000-1000-8000-00805f9b34fb, id: 21, value: 2034332030302039)
06-17 21:25:25.261 31817 31832 D RxBle#BluetoothGatt: onCharacteristicChanged characteristic=00002af0-0000-1000-8000-00805f9b34fb
06-17 21:25:25.262  6651  6792 E bt_btif : bta_gattc_process_indicate, ignore HID ind/notificiation
06-17 21:25:25.263 31817  1997 V RxBle#BleModule: Notification from Characteristic(uuid: 00002af0-0000-1000-8000-00805f9b34fb, id: 21, value: 32200D)
06-17 21:25:25.263 31817 31831 D RxBle#BluetoothGatt: onCharacteristicChanged characteristic=00002af0-0000-1000-8000-00805f9b34fb
06-17 21:25:25.265 31817  1997 V RxBle#BleModule: Notification from Characteristic(uuid: 00002af0-0000-1000-8000-00805f9b34fb, id: 21, value: 313A203130203433)
06-17 21:25:25.277  6651  6792 E bt_btif : bta_gattc_process_indicate, ignore HID ind/notificiation
06-17 21:25:25.278  6651  6792 E bt_att  : invalid response/indicate pkt size: 31, PDU size: 23
06-17 21:25:25.279 31817 31889 D RxBle#BluetoothGatt: onCharacteristicChanged characteristic=00002af0-0000-1000-8000-00805f9b34fb
06-17 21:25:25.280 31817  1997 V RxBle#BleModule: Notification from Characteristic(uuid: 00002af0-0000-1000-8000-00805f9b34fb, id: 21, value: 20323020343320333020343320343020)
06-17 21:25:25.298  6651  6792 E bt_att  : invalid response/indicate pkt size: 34, PDU size: 23
06-17 21:25:25.298  6651  6792 E bt_btif : bta_gattc_process_indicate, ignore HID ind/notificiation
06-17 21:25:25.300 31817 31832 D RxBle#BluetoothGatt: onCharacteristicChanged characteristic=00002af0-0000-1000-8000-00805f9b34fb
06-17 21:25:25.302 31817  1997 V RxBle#BleModule: Notification from Characteristic(uuid: 00002af0-0000-1000-8000-00805f9b34fb, id: 21, value: 20343320423020343320433020343320)
06-17 21:25:25.315  6651  6792 E bt_btif : bta_gattc_process_indicate, ignore HID ind/notificiation
06-17 21:25:25.317 31817 31831 D RxBle#BluetoothGatt: onCharacteristicChanged characteristic=00002af0-0000-1000-8000-00805f9b34fb
06-17 21:25:25.318 31817  1997 V RxBle#BleModule: Notification from Characteristic(uuid: 00002af0-0000-1000-8000-00805f9b34fb, id: 21, value: 0D)
06-17 21:25:25.334  6651  6792 E bt_att  : invalid response/indicate pkt size: 53, PDU size: 23
06-17 21:25:25.353  6651  6792 E bt_btif : bta_gattc_process_indicate, ignore HID ind/notificiation
06-17 21:25:25.357 31817 31889 D RxBle#BluetoothGatt: onCharacteristicChanged characteristic=00002af0-0000-1000-8000-00805f9b34fb
06-17 21:25:25.360 31817  1997 V RxBle#BleModule: Notification from Characteristic(uuid: 00002af0-0000-1000-8000-00805f9b34fb, id: 21, value: 0D3E)

Cierpliwy, 2017-06-17:
`bt_att` and `bt_btif` errors are definitely Android ones and not related to our library. I can only speculate that Android is not accepting messages larger then negotiated MTU (23 in your case which is minimal value). In that case it might be an implementation error on peripheral side.

hhtu, 2017-06-17:
Yes, I know that, but the weird part is, sometimes monitorCharacteristicForDevice gets data correctly, sometimes doesn't, and when I use another ODB2 app (Android), it can get data correctly, so I don't know how to trace it deeper to solve this problem.

hhtu, 2017-06-20:
Hi, it seems android has 20 bytes limit for MTU, and that's why I keep failing.

After several googling, I notice this library use RxAndroidBle for android, and when I checked RxAndroidBle page, there is a Long WRITE function 

`
device.establishConnection(false)
    .flatMap(rxBleConnection -> rxBleConnection.createNewLongWriteBuilder()
        .setCharacteristicUuid(uuid) // required or the .setCharacteristic()
        // .setCharacteristic() alternative if you have a specific BluetoothGattCharacteristic
        .setBytes(byteArray)
        // .setMaxBatchSize(maxBatchSize) // optional -> default 20 or current MTU
        // .setWriteOperationAckStrategy(ackStrategy) // optional to postpone writing next batch
        .build()
    )
    .subscribe(
        byteArray -> {
            // Written data.
        },
        throwable -> {
            // Handle an error here.
        }
    );
`

 I thought maybe I can change the default MTU size bigger for solving my problem, but I'm not sure if you put this in react-native-ble-plx? because I searched for several key terms but found nothing.

Sorry to bother again.

Cierpliwy, 2017-06-20:
Sorry for responding so late.

I agree that trying negotiating bigger MTU value may fix your issues. I'll try to add this option to a library tomorrow when I will have BLE devices in my office.

Cierpliwy, 2017-06-21:
Hello,

Could check temporary implementation which allows you to request MTU during device connection? It is available on branch request_mtu, which you can use by adding this line to `package.json`:
`"react-native-ble-plx": "git://github.com/Polidea/react-native-ble-plx.git#request_mtu"`

When connecting to device you can pass optional connection arguments:
`device.connect({requestMtu: 50})`

Then search for following lines in logs:
```
D/BluetoothGatt: onConfigureMTU() - Device=B4:99:4C:34:DE:77 mtu=23 status=0
D/RxBle#BluetoothGatt: onMtuChanged mtu=23 status=0
```
In my test device I was not able to get higher MTU then 23.

hhtu, 2017-06-21:
Oh my god, it's work!! Not only 50, I even changed it to 100, because my simulator will return more than 70, and I received all of them, not a single one left!!!!

And my adb log did not show any error messages like before.
```
06-21 23:59:02.077  7902  8013 D bt_btm  : btm_bda_to_acl found
06-21 23:59:02.077  7902  7919 I bt_btif : HAL bt_gatt_callbacks->client->notify_cb
```

Although I don't know if there is a maximum value or not.

But anyway, thank you so much~~!!!!!!

Maybe you are the first react-native bluetooth library support this feature, because I also tried others, all of them have this issue....

You save my day, thank you again!!

Cierpliwy, 2017-06-21:
No problem. You can use `device.getMtuForDevice()` (bad name from my side) to check what negotiated value really is. I will try to clean API, add tests, add support for iOS and then release 0.7.0 version with new feature. 

Probably I will change function signatures as well.

Cierpliwy, 2017-06-21:
I will close it when feature is completed ;)

andycamp, 2017-10-11:
Hi, We're interested in this feature for the same reason. Any idea when this will be finished? Thanks.

hhtu, 2017-10-20:
Hi again

Recently I update my mac to Xcode 9, and I'm facing the Xcode 9 compile Rx problem in "react-native-ble-plx": "git://github.com/Polidea/react-native-ble-plx.git#request_mtu"

I know this problem has been resolved in 0.6.2, but I can't use this one.

When will you merge this into standard version? or can you help me to solve xcode 9 in this request_mtu version?

Thanks.

Cierpliwy, 2017-10-24:
Hello

Version 0.6.4 added `mtu` property to `Device` object which allows you to read it after device is connected with `requestMTU` option specified or by calling `requestMTUForDevice` function.  On iOS requesting MTU is noop operation as iPhone is doing it automatically (however after a while when device is connected it should be available and in most cases it is 158 bytes). On Android make sure to call one of above functions.

closed 2018-04-09 by Cierpliwy