dotintent/react-native-ble-plx #93: Error instantiating BleManager (undefined is not an object)

erhathaway, 2017-06-20:
I am getting the error `undefined is not an object (evaluating '_BleModule.BleModule.createClient')`

I'm importing the library:
```
import { BleManager } from 'react-native-ble-plx'
```

And instantiating the manager in the constructor:
```
  constructor() {
      super();
      this.manager = new BleManager();
  }
``

I ejected from CRNA and ran 
```
$ npm install --save react-native-ble-plx
$ react-native link
```

My package.json is:
```
{
  "name": "mobile-client",
  "version": "0.1.0",
  "private": true,
  "devDependencies": {
    "babel-eslint": "^7.2.3",
    "enzyme": "^2.8.2",
    "eslint": "^3.19.0",
    "eslint-config-airbnb": "^15.0.1",
    "eslint-plugin-flowtype": "^2.34.0",
    "eslint-plugin-import": "^2.2.0",
    "eslint-plugin-jsx-a11y": "^5.0.3",
    "eslint-plugin-react": "^7.0.1",
    "eslint-plugin-react-native": "^2.3.2",
    "flow-bin": "^0.40.0",
    "jest-expo": "^0.4.0",
    "jest-serializer-enzyme": "^1.0.0",
    "react-addons-test-utils": "^15.5.1",
    "react-dom": "^15.6.0",
    "react-test-renderer": "^16.0.0-alpha.6",
    "redux-mock-store": "^1.2.3"
  },
  "main": "index.js",
  "scripts": {
    "test": "node node_modules/jest/bin/jest.js --watch",
    "flow": "flow",
    "lint": "node node_modules/eslint/bin/eslint.js -c ./.eslintrc.json --ignore-pattern '*/node_modules/*' --ignore-pattern '*/npm/*' --ignore-pattern '*/coverage/*' .; exit 0"
  },
  "jest": {
    "preset": "jest-expo",
    "transformIgnorePatterns": [
      "node_modules/(?!react|@expo|expo)"
    ],
    "snapshotSerializers": [
      "./node_modules/jest-serializer-enzyme"
    ]
  },
  "dependencies": {
    "apollo-client": "^1.4.0",
    "expo": "^16.0.0",
    "prop-types": "^15.5.10",
    "react": "16.0.0-alpha.6",
    "react-apollo": "^1.4.0",
    "react-native": "https://github.com/expo/react-native/archive/sdk-16.0.0.tar.gz",
    "react-native-ble-plx": "^0.6.0",
    "react-navigation": "^1.0.0-beta.11",
    "react-redux": "^5.0.5",
    "redux": "^3.6.0",
    "redux-persist": "^4.8.0",
    "redux-saga": "^0.15.3",
    "redux-thunk": "^2.2.0"
  }
}
```

My AndroidManifest.xml looks like:
```
<?xml version="1.0" encoding="utf-8"?>
<manifest package="mobile-client"
          xmlns:android="http://schemas.android.com/apk/res/android">

  <uses-permission android:name="android.permission.BLUETOOTH"/>
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
  <uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION"/>
  <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>

  <application
    android:name=".MainApplication"
    android:allowBackup="true"
    android:icon="@mipmap/ic_launcher"
    android:label="@string/app_name"
    android:supportsRtl="true">
    <activity android:name=".MainActivity"
              android:configChanges="keyboard|keyboardHidden|orientation|screenSize"
              android:theme="@style/Theme.Exponent.Light"
              android:windowSoftInputMode="adjustResize">
      <intent-filter>
        <action android:name="android.intent.action.MAIN"/>

        <category android:name="android.intent.category.LAUNCHER"/>
      </intent-filter>

      <intent-filter>
        <data android:scheme="exp7ddd4f484fc44690a39d9fc9b751f23d"/>

        <action android:name="android.intent.action.VIEW"/>

        <category android:name="android.intent.category.DEFAULT"/>
        <category android:name="android.intent.category.BROWSABLE"/>
      </intent-filter>
    </activity>
  </application>

</manifest>
```

build.gradle:
```
apply plugin: 'com.android.application'

android {
  compileSdkVersion 24
  buildToolsVersion '24.0.3'

  defaultConfig {
    applicationId "mobile-client"
    minSdkVersion 19
    targetSdkVersion 24
    versionCode 1
    versionName "1.0"
    multiDexEnabled true
    ndk {
      abiFilters 'armeabi-v7a', 'x86'
    }
    manifestPlaceholders = [
      'appAuthRedirectScheme': 'mobile-client'
    ]
  }
  buildTypes {
    release {
      minifyEnabled false
      proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
    }
  }
  dexOptions {
    javaMaxHeapSize "8g"
  }
}

task exponentPrebuildStep(type: Exec) {
  workingDir '../../'

  if (System.getProperty('os.name').toLowerCase().contains('windows')) {
    commandLine 'cmd', '/c', '.\\.expo-source\\android\\detach-scripts\\prepare-detached-build.bat'
  } else {
    commandLine './.expo-source/android/detach-scripts/prepare-detached-build.sh'
  }
}
preBuild.dependsOn exponentPrebuildStep

repositories{
  flatDir{
    dirs 'libs'
  }
  mavenLocal()
  maven { url 'https://maven.fabric.io/public' }
}

dependencies {
    compile project(':react-native-ble-plx')
  compile fileTree(dir: 'libs', include: ['*.jar'])
  testCompile 'junit:junit:4.12'
  compile 'com.android.support:appcompat-v7:24.1.1'
  compile 'com.android.support:multidex:1.0.0'

  
  compile('host.exp.exponent:expoview:16.0.0@aar') {
    exclude group: 'com.facebook.android', module: 'facebook-android-sdk'
    exclude group: 'com.facebook.android', module: 'audience-network-sdk'
    exclude group: 'io.nlopez.smartlocation', module: 'library'
    transitive = true;
  }

  compile ('com.facebook.android:facebook-android-sdk:4.7.0') {
    exclude module: 'bolts-android'
  }
  compile('com.facebook.android:audience-network-sdk:4.19.0') {
    exclude module: 'play-services-ads'
  }
  compile('io.nlopez.smartlocation:library:3.2.11') {
    transitive = false
  }
  
  
}
```

MainApplication.java:
```
package mobile-client;

import android.support.multidex.MultiDexApplication;

import com.facebook.react.ReactPackage;

import java.util.Arrays;
import java.util.List;

// Needed for `react-native link`
// import com.facebook.react.ReactApplication;
import com.polidea.reactnativeble.BlePackage;

public class MainApplication extends MultiDexApplication {

  // Needed for `react-native link`
  public List<ReactPackage> getPackages() {
    return Arrays.<ReactPackage>asList(
        // Add your own packages here!
        // TODO: add cool native modules

        // Needed for `react-native link`
        // new MainReactPackage(),
            new BlePackage()
    );
  }
}
```

erhathaway, 2017-06-22:
This issue was caused by a problem with packaging the app. Closing it

closed 2017-06-22 by erhathaway

batusai513, 2017-07-18:
@erhathaway hi, I wanted to know how did you fixed that, I'm having the same problem, thanks in advanced.

erhathaway, 2017-07-18:
@batusai513 I didn't document how I fixed it at the time, but I was having issues with how android was building the app (Build > Make Project in Android Studio). Sorry I don't have more info to offer. 

Are you able to build the project and get it on the phone successfully?

JordanKlaers, 2017-10-21:
I am also having the same issue and the project does build and pulls up on my phone through Expo XDE, and just like you when I call the constructor it fails the same, any suggestions you can provide?

skrichten, 2018-01-23:
Same issue as OP here.  I ejected a blank app from CRNA