dotintent/react-native-ble-plx #94: Calling destory method throws unhandled Promise rejection.

Elektro1776, 2017-06-21:
Currently we have a global BLE Manager that is handling all of our connections , and when we would like to destroy every connection existing we experience a unhandled Promise Rejection with the message , "Error:Error:Destroyed". From the docs and the recent update on 0.6.0 it seems like all promises and subscriptions in waiting should be destroyed properly now. Anyone with some possible insight into why we would experience this issue would be greatly appreciated ! Other than that freaking love this library, Cheers!

Cierpliwy, 2017-06-21:
This message looks strange. After calling `destroy()` function all pending promises should return error with message equal to "Destroyed" for now. What RN version are you using and on what platform you are receiving this message?

Elektro1776, 2017-06-21:
Verions: 
         "react": "16.0.0-alpha.12",
        "react-native": "0.45.1",
And we receive this message on both Android and IOS

GroupXTech, 2017-06-22:
I have the same issue, 
"Possible Unhandled Promise Rejection (id: 0):
Error: Error: Destroyed"
Both Android and IOS, RN-0.45.1
Nothing special in my code -  manager.destroy();

RangerMauve, 2017-09-07:
@GroupXTech The issue there is you're not invoking `.catch` on promises returned from the various methods, therefor when they reject it throws an "Unhandled Promise Rejection".

The solution is to make sure to add catch handlers to all your promises from the lib

GroupXTech, 2017-09-07:
Ahhh, Yes I do tend to be lazy about that.

Cierpliwy, 2018-04-09:
Closing due to inactivity.

closed 2018-04-09 by Cierpliwy

lxcool9243, 2019-11-25:
> Ahhh, Yes I do tend to be lazy about that.

Can you please ellaborate how you resolved the issue as i am having same issues.

ayberkgerey, 2021-02-11:
@lxcool9243 Did you solve it ? because i have the same problem.