dotintent/react-native-ble-plx #96: Can't get all infos about devices

OlivierVillequey, 2017-06-26:
Hi! As a first goal, I wanted to get a list of all devices in my surroundings with the basic functions you detail in your documentation. However, I have an issue to get all the datas about my devices even though the function seems to work fine. I get a device object like this:

Device {
   id: "2F:AE:4D:B6:C4:3D",
   isConnectable: null,
   manufacturerData: "BgABCSAA18c+9sBY2xXpcA/XVZEUEtFD1uZIHQ8=↵",
  name: null,
   overflowServiceUUIDs: null,
   rssi: -37,
   serviceData: null,
   serviceUUIDs: null,
   solicitedServiceUUIDs: null,
   txPowerLevel: null,
}

I can understand that we need to connect to it to retrieve some datas like services but for example why the name is null? Something is wrong here and I don't get it.
I'm sorry, I'm new to react native and handling bluetooth between peripherals.
I already block on something that simple when my final goal is to see whether or not we can configure the connected devices wifi using bluetooth.

Cierpliwy, 2017-06-26:
During scanning you can only see limited information about device. There is special packet called Advertisement Data which is used to send above information. It is limited in size (max 29 bytes of user payload) and as I can see all of it is used by manufacturerData in your case.

After connection you can try to read name from characteristic if it's available:
https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.gap.device_name.xml

OlivierVillequey, 2017-06-26:
Hi, thank you for you quick reply.
I see what's the problem, so how can I ignore manufacturerData to get other informations that I need? 
Maybe there is a way to loop on this advertisement Data call to get the whole thing?

Thanks.

Cierpliwy, 2017-06-26:
Well you can't do anything from application side. Peripheral needs to be reconfigured / its firmware modified.

guerjon, 2017-08-16:
@Cierpliwy  There is any way to get the ManufacturerSpecificData from the device? 
Actually, i just can get the ManufacturerData, but not the specific, I review your code and you use the ScanRecord class to get the bytes from the manufacturerData but you never call to the specificManufacturerData method and I believe that is important in a BLE library.

closed 2019-03-23 by stale[bot]