dotintent/react-native-ble-plx #97: Access BleClientManager instance

patlux, 2017-06-26:
Hello,

is it possibe from the native side to get the instance of the BleClientManager?

I'm scanning with your module for bluetooth devices and then I'm passing (like in your javascript) the `deviceIdentifier` to **my** native module. Like in your library I need now access to the `connectedPeripherals`, which is defined in the BleClientManager class. So how I get the Instance of BleClientManager from my native module in Swift (iOS)?

On android I have found a workaround: `BluetoothAdapter.getDefaultAdapter().getRemoteDevice(deviceIdentifier)`.
For iOS I have found nothing similiar.

Cierpliwy, 2017-06-26:
Do you need access to BleClientManager (react-native-ble-plx) or  BluetoothManager (RxBluetoothKit) or CBCentralManager (iOS)? Each abstraction level don't give you access to its internals.

patlux, 2017-06-26:
I need access to the BleClientManager (react-native-ble-plx).

Cierpliwy, 2017-07-03:
In that case you probably can investigate how native modules are registered in RN and used by it. I bet it will be very troublesome and prone to internal changes in RN native library. Therefore I suggest to fork this library and add your own "gateway" to created BleClientManager object by modifying BleClient Objective-C static library.

closed 2019-03-23 by stale[bot]