dotintent/react-native-ble-plx #98: 'react-native link' unrecognized

rpiwetz, 2017-06-27:
Started with a new project:
`react-native init appName`
`cd appName`
Verified that `react-native link` works.  Successful
Added react-native-ble-plx:
`npm install --save react-native-ble-plx`
Tried `react-native link`, receive error:
"Command `link` unrecognized. Make sure that you have run `npm install` and that you are inside a react-native project."

"dependencies": {
		"react": "16.0.0-alpha.12",
		"react-native": "0.45.1",
		"react-native-ble-plx": "^0.6.0"
}

Cierpliwy, 2017-06-29:
I did the same thing as you and `react-native link` command resulted in:
```
Scanning 558 folders for symlinks in /Users/cierpliwy/git/appName/node_modules (5ms)
rnpm-install info Linking react-native-ble-plx android dependency 
rnpm-install info Android module react-native-ble-plx has been successfully linked 
rnpm-install info Linking react-native-ble-plx ios dependency 
rnpm-install info iOS module react-native-ble-plx has been successfully linked
```

inev, 2017-06-29:
I have the same issue. I was able to get around it by deleting `node_modules` and typing `yarn`.

rpiwetz, 2017-07-13:
@inev, thank you, that fixed it for me.  Tried with brand new RN app again and ran into the issue, so deleted node_modules and ran yarn...so had to sit through the building of all those framework files twice, but at least it works now.

sakettawde, 2017-08-26:
Thanks, @inev had the same problem, repeated the same steps, deleted node_modules folder and ran 'yarn' command and then ran 'react-native link', seemed to work then.

hhff, 2017-09-07:
same!  yarn FTW~

jonrh, 2017-10-25:
Fell victim to this. My mistake was that I ran `react-native link` inside the wrong folder. I was in the `android/` folder. For `react-native link` to work you need to be located at the root of your React Native project.

This is because the command `react-native` only has the command `init` available globally. All other commands, for example `link`, rely on the React Native version you have installed in `node_modules`.

Cierpliwy, 2017-10-28:
I'll add note to README file about it. I was also confused about it.

volkansahin45, 2017-11-29:
Make sure your folder name and package.json name are equals. I used that mistake, I accidently changed the name in the package.json

Cierpliwy, 2017-11-29:
About which folder name are you talking about @volkansahin45?

volkansahin45, 2017-11-29:
@Cierpliwy the application folder

sangampandey, 2017-12-15:
I believe this happens after you add any new library merely run yarn install after adding any library and later react-native link should definitely work

rastogitech, 2018-01-02:
`react-native link` can not be used in "create-react native-app" projects.

mayankzen123, 2018-01-02:
npm install native-base --save
npm install
react-native link

This sequence worked for me

rastogitech, 2018-01-03:
Did it work in expo un-detached project?

naxir, 2018-01-04:
@sangampandey thanks your solution worked for me

bashirpour, 2018-05-06:
i run the `npm install`
and my problem was solved

zero1world, 2018-11-19:
no one solution is running for me

ZeldaYuan, 2018-12-06:
Me too, none of them works for me.

TheBlackHacker, 2018-12-10:
`npm install` inside the project folder solved the problem.

SL-A-SH, 2019-01-31:
No of the solutions work for either. 

- Removed node_modules folder.
- Installed yarn
- Typed 'yarn' and then 'react-native link react-native-vector-icons'

Still got the error **_Command `link` unrecognized. Make sure that you have run `npm install` and that you are inside a react-native project._**

I am on a Linux machine.

closed 2019-04-08 by stale[bot]

Usamaliaquat123, 2019-04-24:
this no solution worked for me ...!

Memexp, 2019-04-25:
Me neither, it still says:

bash: react-native: command not found

shusain, 2019-08-01:
@Memexp you need to install react-native globally for the command to be found in your path `npm install -g react-native` or `yarn add global react-native`

rinardmorais, 2019-08-25:
npm install react-native-firebase work!

shusain, 2019-08-29:
Correction on what I said above (will edit as well) the globally installed package should be `react-native-cli` not `react-native`( this one should just be installed with a regular npm install in your project not globally) this will make the `react-native` commands available to use anywhere (assuming node module globals bin is in your system path)

kest-cloud, 2019-08-29:
it really worked.....remove node_modules and run Yarn....worked perfectly. thanks so much

daminduLiyanage, 2019-09-28:
This should work: 

`npm install --save react-native-firebase`
`react-native link react-native-firebase`

matheusfaggi, 2020-01-20:
I try remove node_modules then run yarn, but does not work. I was adding dependencies directly on package.json. Now i try add each dependecies using yarn add, and work it.

zacdemi, 2021-01-26:
In my case I was using npx create-react-app my-app instead of npx create-react-native-app

kirbi96, 2022-06-30:
if you want add fonts, for react native 0.69 run
npx react-native-asset

Zontex, 2022-07-03:
> if you want add fonts, for react native 0.69 run npx react-native-asset

Saved my life.

TravisHi, 2022-07-20:
> if you want add fonts, for react native 0.69 run npx react-native-asset

Do we still have to configure the `react-native.config.js`?

mothy900, 2022-09-11:
> if you want add fonts, for react native 0.69 run npx react-native-asset

Saved my life 2222 Thanks!!!!!!!

Zhouwenli, 2023-04-06:
> `react-native link`不能用于“create-react native-app”项目。

为什么？

hannan228, 2023-05-21:
no need for react-native link now
[https://reactnative.dev/blog/2022/09/05/version-070#android-auto-linking-for-new-architecture-libraries](android auto link update)