dotintent/react-native-ble-plx #100: writeCharacteristicWithResponseForService promise never gets called

watadarkstar, 2017-06-29:
I am calling `writeCharacteristicWithResponseForService` and the response / promise never gets called. It never prints out `console.log('characteristic:');`. I don't know what I am doing wrong...

```javascript
device.writeCharacteristicWithResponseForService(
            BATTERY_SERVICE_UUID, BATTERY_CHARACTERISTIC_UUID, valueBase64, transactionId
        ).then(characteristic => {
            console.log('characteristic:'); // < ---- never gets called
            console.log(characteristic);
        }).catch(error => {
            console.log('error');
            console.log(error);
        });
```

 I use the [Anaren](https://atmosphere.anaren.com/wiki/Main_Page) for my bluetooth chip. I am seeing the logs when I call `writeCharacteristicWithResponseForService` . It does print `readBattery`, but as I said above there is no response callback.

```C
char* readBattery(char* data, int length) {
    print("readBattery()\r\n");
    setLedColor(GREEN);
    
	static char buffer[MAX_LEN];
	snprintf(buffer,MAX_LEN,'dan');
	
	return buffer;
}
```

Not sure if this is a problem on `react-native-ble-plx` or my embedded code... Any ideas? 

Thanks

watadarkstar, 2017-06-30:
@Cierpliwy Please see the above and below I would really appreciate any help.

Does `writeCharacteristicWithResponseForService perform a `writeCharacteristicForService` followed by a `readCharacteristicForService`. Because I can't get `writeCharacteristicWithResponseForService` to work but if I make two buttons one that calls `writeCharacteristicForService` and then one that calls `readCharacteristicForService` then things work. And click them in that order it works, I get a read promise callback. But `writeCharacteristicWithResponseForService` does not work, the promise callback is never called...

watadarkstar, 2017-06-30:
I ended up using `monitorCharacteristicForService`. However, its still odd that the callback is never called for `writeCharacteristicWithResponseForService`...

Cierpliwy, 2017-07-03:
The best way to monitor what is happening on mobile application boundary is to enable logging via:
https://polidea.github.io/react-native-ble-plx/#BleManager#setLogLevel, and then observe them in XCode/AndroidStudio to see what is happening. Alternatively you can use other application to test BLE communication like nRF Connect.

watadarkstar, 2017-07-06:
@Cierpliwy Thank you!

Cierpliwy, 2017-07-07:
@watadarkstar Did you discover any reason why `writeCharacteristicWithResponseForService` didn't resolve?

watadarkstar, 2017-07-12:
@Cierpliwy No I ended up using `monitorCharacteristicForService`.

closed 2017-07-13 by Cierpliwy

ArthurRuxton-DY, 2022-10-12:
@watadarkstar - hi, monitorCharacteristicForService doesn't take a base64 arg, how did you get around this in the end?