dotintent/react-native-ble-plx #99: Does example work on Ubuntu?

inev, 2017-06-29:
I went into the example, typed `yarn` and then typed `npm start` and i got:
```
> ReactBLEScanner@0.0.1 start /home/aaa/Downloads/react-native-ble-plx-master/examples/ReactBLEScanner
> node node_modules/react-native/local-cli/cli.js start

/home/aaa/Downloads/react-native-ble-plx-master/examples/ReactBLEScanner/node_modules/babel-core/lib/transformation/file/options/option-manager.js:328
        throw e;
        ^

Error: Couldn't find preset "react-native" relative to directory "/home/aaa/Downloads/react-native-ble-plx-master"
    at /home/aaa/Downloads/react-native-ble-plx-master/examples/ReactBLEScanner/node_modules/babel-core/lib/transformation/file/options/option-manager.js:293:19
    at Array.map (native)
    at OptionManager.resolvePresets (/home/aaa/Downloads/react-native-ble-plx-master/examples/ReactBLEScanner/node_modules/babel-core/lib/transformation/file/options/option-manager.js:275:20)
    at OptionManager.mergePresets (/home/aaa/Downloads/react-native-ble-plx-master/examples/ReactBLEScanner/node_modules/babel-core/lib/transformation/file/options/option-manager.js:264:10)
    at OptionManager.mergeOptions (/home/aaa/Downloads/react-native-ble-plx-master/examples/ReactBLEScanner/node_modules/babel-core/lib/transformation/file/options/option-manager.js:249:14)
    at OptionManager.init (/home/aaa/Downloads/react-native-ble-plx-master/examples/ReactBLEScanner/node_modules/babel-core/lib/transformation/file/options/option-manager.js:368:12)
    at compile (/home/aaa/Downloads/react-native-ble-plx-master/examples/ReactBLEScanner/node_modules/babel-register/lib/node.js:103:45)
    at loader (/home/aaa/Downloads/react-native-ble-plx-master/examples/ReactBLEScanner/node_modules/babel-register/lib/node.js:144:14)
    at Object.require.extensions.(anonymous function) [as .js] (/home/aaa/Downloads/react-native-ble-plx-master/examples/ReactBLEScanner/node_modules/babel-register/lib/node.js:154:7)
    at Module.load (module.js:503:32)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! ReactBLEScanner@0.0.1 start: `node node_modules/react-native/local-cli/cli.js start`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the ReactBLEScanner@0.0.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

```

Cierpliwy, 2017-07-03:
Example is currently outdated and it needs more work to support latest version of this library and RN.

inev, 2017-07-03:
@Cierpliwy Is there a working project I can look at to see the steps needed to make it work with the current versions? Thank You.

CandyOgre, 2017-08-02:
@Cierpliwy so, how can we start example? Very very needed

closed 2019-03-23 by stale[bot]