dotintent/react-native-ble-plx #101: huge delay in writing characteristics

gvaldambrini, 2017-07-05:
Hi,
I'm trying to read & write a set of characteristics from / to a device. While reading from the device just works fine, every time I try to write a characteristic the value is updated after a huge amount of time (greater than 20 secs in the best case). 
I am on iOS and I already checked if the device works fine using the BleScanner app for which everything works as expected (the value is immediately updated).

As the characteristic`isWritableWithResponse`, I used the following code to write:
```
    console.log('write to device');
    characteristic.writeWithResponse(hexToBase64(value)).then(
      () => console.log('write response'));
```
The value is updated only when the promise is resolved:
```
2017-07-05 12:41:13.323 [info][tid:com.facebook.React.JavaScript] write to device
2017-07-05 12:41:33.206 [info][tid:com.facebook.React.JavaScript] write response
```
Am I missing something? The whole component (which reads in polling the characteristics from the device and allows to change one of those) can be found here: https://gist.github.com/gvaldambrini/2ead28fbff8d27cf2047d9011e04f14a

Thanks!

Cierpliwy, 2017-07-05:
Please enable native module's log:
https://polidea.github.io/react-native-ble-plx/#BleManager#setLogLevel
and run your application from XCode to see when exactly iOS received callback with successful write.

gvaldambrini, 2017-07-05:
Activating the native module's log seems that the problem is reading all the 9 characteristics in a polling interval of 1sec as I can see from the Xcode log that each read on the device takes ~300ms:
```
[RxBLEKit|DEBG|12:46:36.646]: Peripheral(uuid: E719CA00-6B55-47FC-90C2-71E48D7D591F, name: Optional("MyDevice")) didUpdateValueFor(for:Characteristic(uuid: 334869D3-8DBA-47AC-9B8E-B5BC3ADF1727, id: 6174764928), value: Optional("deadbeef"), error: nil)
[RxBLEKit|DEBG|12:46:36.931]: Peripheral(uuid: E719CA00-6B55-47FC-90C2-71E48D7D591F, name: Optional("MyDevice")) didUpdateValueFor(for:Characteristic(uuid: 334869D3-8DBA-47AC-9B8E-B5BC3ADF1728, id: 6174763296), value: Optional("42"), error: nil)
[RxBLEKit|DEBG|12:46:37.216]: Peripheral(uuid: E719CA00-6B55-47FC-90C2-71E48D7D591F, name: Optional("MyDevice")) didUpdateValueFor(for:Characteristic(uuid: 334869D3-8DBA-47AC-9B8E-B5BC3ADF1729, id: 6174764352), value: Optional("aa"), error: nil)
[RxBLEKit|DEBG|12:46:37.501]: Peripheral(uuid: E719CA00-6B55-47FC-90C2-71E48D7D591F, name: Optional("MyDevice")) didUpdateValueFor(for:Characteristic(uuid: 334869D3-8DBA-47AC-9B8E-B5BC3ADF172A, id: 6174764448), value: Optional("abadcafe"), error: nil)
[RxBLEKit|DEBG|12:46:37.786]: Peripheral(uuid: E719CA00-6B55-47FC-90C2-71E48D7D591F, name: Optional("MyDevice")) didUpdateValueFor(for:Characteristic(uuid: 334869D3-8DBA-47AC-9B8E-B5BC3ADF172B, id: 6174764064), value: Optional("c001babe"), error: nil)
[RxBLEKit|DEBG|12:46:38.072]: Peripheral(uuid: E719CA00-6B55-47FC-90C2-71E48D7D591F, name: Optional("MyDevice")) didUpdateValueFor(for:Characteristic(uuid: 334869D3-8DBA-47AC-9B8E-B5BC3ADF172C, id: 6174763680), value: Optional("01020304"), error: nil)
[RxBLEKit|DEBG|12:46:38.356]: Peripheral(uuid: E719CA00-6B55-47FC-90C2-71E48D7D591F, name: Optional("MyDevice")) didUpdateValueFor(for:Characteristic(uuid: 334869D3-8DBA-47AC-9B8E-B5BC3ADF172D, id: 6174763968), value: Optional("fe"), error: nil)
```
Increase the polling interval to something like 10sec makes the write fast as with BleScanner so I imagine that the right way of doing it would be using the BLE monitor capabilities. 
Thanks for the help!

closed 2017-07-05 by gvaldambrini