dotintent/react-native-ble-plx #102: Android ask for permissions at runtime Android M (API 23+)

watadarkstar, 2017-07-06:
I use this library in our app.

I have this issue In Android M (API 23+) where you are suppose to request permissions at runtime. So I'm suppose to implement React Native's [`PermissionAwareActivity`](https://github.com/facebook/react-native/blob/master/ReactAndroid/src/main/java/com/facebook/react/modules/core/PermissionAwareActivity.java). And use [PermissionsAndroid](https://facebook.github.io/react-native/docs/permissionsandroid.html) to request permissions. My code looks like this:

```
async checkAndGrantPermissions() {
        console.log('checkAndGrantPermissions');
        try {
            console.log('try');
            const granted = await PermissionsAndroid.request(
                PermissionsAndroid.PERMISSIONS.BLUETOOTH,
                {
                title: 'Cool App Bluetooth Permission',
                message: 'Cool App needs access to your bluetooth ' +
                'so you can be awesome.'
                }
            );
            console.log('granted');
            console.log(granted);
        } catch (err) {
            console.log('errrr');
            console.log(err);
        }
    }
```

But I get `permission is null` when I use `PermissionsAndroid.PERMISSIONS.BLUETOOTH`. It seems like `PermissionsAndroid.PERMISSIONS.BLUETOOTH` doesn't exist?

watadarkstar, 2017-07-06:
@Cierpliwy was wondering if you knew anything about this? Any help would be appreciated.

MBach, 2017-07-07:
@watadarkstar you should try another permission :

    constructor(props) {
        super(props)
        if (Platform.OS === 'android') {
            PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION)
            .then(granted => {
                if (granted !== 'granted') {
                    console.log('BLUETOOTH denied')
                }
            })
        }
    }

watadarkstar, 2017-07-07:
@MBach That works but I need bluetooth permissions to be accepted. Does `ACCESS_COARSE_LOCATION` give me bluetooth permissions?

watadarkstar, 2017-07-07:
That worked @MBach thank you 👍

closed 2017-07-13 by Cierpliwy