dotintent/react-native-ble-plx #104: manager - detect peripheral packets after disconnect

brycejacobs, 2017-07-13:
Has anyone else had this problem? I'm trying to disconnect from a peripheral and pick it up while scanning again.

For whatever reason, the packets aren't being detected. Any idea what's going on?

React-Native: 0.44
Node: 8.1.3

brycejacobs, 2017-07-13:
Hm, I added a delay between stopping the scan, and starting up a new scan, and it's working again.

So for whatever reason, if you start the scan too fast from stopping it, it will not detect packets from a device that's been connected to.

Will try to whip up something reproducible for you guys.

samthui, 2018-01-03:
Seems like my issue here #198.

I guess the problem is due to the peripheral's state of connection. Should manage it in the Device class itself?

Cierpliwy, 2018-01-03:
@brycejacobs This issue should be fixed with version 0.7. There was iOS bug related to restarting scanning.

closed 2019-03-23 by stale[bot]