dotintent/react-native-ble-plx #105: Module 'BleClientManager' not found when using `yarn add react-native-ble-plx`

tomdaniel0x01, 2017-07-14:
The Carthage folder does not show up in $(SRCROOT)/../node_modules/react-native-ble-plx/ios/BleClientManager/ when using yarn.

Cierpliwy, 2017-07-18:
Probably yarn is not calling postinstall script...

watadarkstar, 2017-07-20:
Are you an expo detached app? Not sure if this is related but you should:

Add Header Build search paths in BLEClient.xcodeproj:
```
${SRCROOT}/../../../ios/Pods/Headers/Public
$(SRCROOT)/../../react-native/React
```

wkoutre, 2017-07-31:
This happens to me sometimes when I `yarn add` a new module. To fix, I simply uninstall a module, which forces the BleClientManager to rebuild itself, after which things return to normal.

tomdaniel0x01, 2017-08-01:
@watadarkstar this was not an Expo app.
@wkoutre I still had this problem even after creating a complete new react-native app through `react-native init`

wkoutre, 2017-08-02:
@tomdaniel0x01 You've deleted your yarn.lock file, node_modules, and did a fresh `yarn install` ?

stinodes, 2017-10-13:
Also ran into this issue.
The post-install script was fired, but didn't complete, logging something about an argument being invalid (something about cache, can't remember exactly why, sorry for that).
Apparently Carthage was out of date. Updating fixed it.

Just a heads-up :)

closed 2019-03-23 by stale[bot]