dotintent/react-native-ble-plx #106: device - rediscover attributes after disconnect?

brycejacobs, 2017-07-18:
Hey guys,

Is there any reason we have to rediscover attributes after a disconnect. Ideally, it would be swell if we could connect to a device, for caching of attributes, then disconnect to free up connections to other devices.

I work for a controls company, and we rotating connections, due to constraints on max allowed connections at any given time. As of right now, any time we want to perform a read write, we have to discover the attributes before hand. 

Any feedback is much appreciated.

Cierpliwy, 2017-07-18:
What do you mean by attributes? Are you concerned that after each connection you are forced to discover services/characteristics again? If so, technically on iOS you can discover only selected services/characteristics before using them, which may be a lot of faster in case you have plenty of them. Unfortunately on Android as far as I know it is not possible. I decided to select most specific implementation for this library and you have to discover everything before use.

brycejacobs, 2017-07-18:
Steps to reproduce:

1. Connect to device
2. Discover attributes (Services, characteristics, etc) for device.

At this point, if I read / write to characteristic, the library acts as expected and performs the procedure.

3. Disconnect Device
4. Re-establish connection to Device
5. Write / Read to characteristic

```bash
Error: Characteristic 67676767-6767-6767-6767-676767670001 not found
    at createErrorFromErrorData (index.ios.bundle:4505)
    at index.ios.bundle:4463
    at MessageQueue.__invokeCallback (index.ios.bundle:4784)
    at index.ios.bundle:4630
    at MessageQueue.__guard (index.ios.bundle:4719)
    at MessageQueue.invokeCallbackAndReturnFlushedQueue (index.ios.bundle:4629)
    at debuggerWorker.js:71
```

Shouldn't the attributes be cached from the connection before? It makes no sense that we would need to rediscover the attributes. From what I remember from my native app days, ios makes use of heavy caching unless CoreBluetooth receives a Changed Service procedure call. (Ref Bluetooth Spec 3.G.7.1)

Cierpliwy, 2017-07-19:
Attributes are cached (_only for paired devices_) from previous connection until peripheral notfies iOS about any attribute changes. Still you have to call `func discoverServices([CBUUID]?)` and `func discoverCharacteristics([CBUUID]?, for: CBService)` before using characteristic.  Although this process should be very fast (almost instantaneous) after second connection due to caching.

Still in this library we are discovering all services and characteristics, which based on number of attributes can take more time then necessary. We could implement lazy discovering for example to mitigate this problem or give users access to finer grained functions.

brycejacobs, 2017-07-19:
I think as long as they're cached on the ios side, it shouldn't be that big of a deal. It's not that big of a deal to have to call one function before doing read / write procedures.

closed 2019-03-23 by stale[bot]