dotintent/react-native-ble-plx #107: discoverAllServicesAndCharacteristics hangs on android

erhathaway, 2017-07-18:
This may be similar to https://github.com/Polidea/react-native-ble-plx/issues/30 but for android. I haven't test it on iOS yet.

Below is my code. I am able to scan and connect to a Polar heart rate monitor but upon running `<device>.discoverAllServicesAndCharacteristics` the method call hangs and never returns a promise.

```
import { BleManager } from 'react-native-ble-plx'
...
  constructor() {
    super()
    this.manager = new BleManager()
  }

  componentWillMount() {
     if (Platform.OS === 'ios') {
       this.manager.onStateChange((state) => {
         if (state === 'PoweredOn') this.scanAndConnect()
       })
     } else {
       this.scanAndConnect()
     }
   }

  scanAndConnect = () => {
    this.manager.startDeviceScan(null, null, (error, device) => {
      console.log("Scanning...")
      console.log(device.name)

      if (error) {
        console.log(error.message)
        return
      }

      if (device && device.name && device.name.includes('Polar')) {
        console.log("Found Polar")

        this.manager.stopDeviceScan()
        this.connectDevice(device)
      }
    });
  }

  async connectDevice(device) {
    const disconnectedDevice = await this.cancelExistingConnection(device)

    console.log("Connecting to Polar")
    const connectedDevice = await disconnectedDevice.connect()

    if (connectedDevice) {
      console.log('Discovering Services and Characteristics')
      const exploredDevice = await connectedDevice.discoverAllServicesAndCharacteristics()
      console.log('Discovering finished')
      await this.listServices(exploredDevice)
      console.log('Listening...')
    } else {
      console.log('failed to connect to device')
    }
  }

  async cancelExistingConnection(device) {
    const isConnected = await device.isConnected()
    console.log(`is connected: ${isConnected}`)

    // return device
    if (isConnected) {
      console.log('canceling existing connection')
      const disconnectedDevice = await device.cancelConnection()
      return disconnectedDevice
    }

    return device
  }

  async listServices(device) {
    console.log(device.serviceData)
    
    const services = await device.services()
    services.forEach(s => {
      console.log(s)
    })
  }
```

Cierpliwy, 2017-07-19:
Thank you for reporting this. Can you enable Verbose logging of native modules and paste registered logs in this issue? You can see them in Android Studio conveniently. 
https://polidea.github.io/react-native-ble-plx/#BleManager#setLogLevel

Today I released v0.6.1, which could potentially fix your problem, make sure to install it earlier.

PS:

You can remove Platform.OS code by calling:
```
this.manager.onStateChange((state) => {
    if (state === 'PoweredOn') this.scanAndConnect()
}, true)
```
https://polidea.github.io/react-native-ble-plx/#BleManager#onStateChange

erhathaway, 2017-07-19:
@Cierpliwy thanks for the quick response and pointer about `onStateChange`!

I updated to 0.6.1, but it didn't solve the issue. It looks like there are two errors now:

```
07-19 10:52:31.028 16373-16856/io.humon.app I/ReactNativeJS: Scanning...
07-19 10:52:31.029 16373-16856/io.humon.app I/ReactNativeJS: Polar H7 B0C74C17
07-19 10:52:31.029 16373-16856/io.humon.app I/ReactNativeJS: Found Polar
07-19 10:52:31.033 16373-16373/io.humon.app D/BluetoothAdapter: stopLeScan()
07-19 10:52:31.033 16373-16373/io.humon.app D/BluetoothAdapter: STATE_ON
07-19 10:52:31.034 16373-16373/io.humon.app D/BluetoothAdapter: STATE_ON
07-19 10:52:31.034 16373-16373/io.humon.app D/BluetoothAdapter: STATE_ON
07-19 10:52:31.035 16373-16373/io.humon.app D/BluetoothAdapter: STATE_ON
07-19 10:52:31.035 16373-16373/io.humon.app D/BluetoothLeScanner: Stop Scan
07-19 10:52:31.037 16373-16856/io.humon.app I/ReactNativeJS: is connected: false
07-19 10:52:31.040 16373-16856/io.humon.app I/ReactNativeJS: Connecting to Polar
07-19 10:52:31.055 16373-16857/io.humon.app D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(8507903)
07-19 10:52:31.055 16373-16917/io.humon.app D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(8507903)
07-19 10:52:31.061 16373-16373/io.humon.app V/RxBle#BleConnectionCompat: Connecting without reflection
07-19 10:52:31.063 16373-16373/io.humon.app D/BluetoothGatt: connect() - device: 00:22:D0:B0:C7:4C, auto: false
07-19 10:52:31.063 16373-16373/io.humon.app D/BluetoothAdapter: isSecureModeEnabled
07-19 10:52:31.063 16373-16373/io.humon.app D/BluetoothGatt: registerApp()
07-19 10:52:31.064 16373-16373/io.humon.app D/BluetoothGatt: registerApp() - UUID=60843038-6a9f-4e7a-b05f-85ce54018b91
07-19 10:52:31.165 16373-16437/io.humon.app D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
07-19 10:52:31.514 16373-16386/io.humon.app D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=00:22:D0:B0:C7:4C
07-19 10:52:31.519 16373-16386/io.humon.app D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=133
07-19 10:52:31.521 16373-16917/io.humon.app D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(8507903)
07-19 10:52:31.523 16373-16386/io.humon.app D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(188299503)
07-19 10:52:31.523 16373-16917/io.humon.app D/RxBle#Radio:  STARTED RxBleRadioOperationDisconnect(188299503)
07-19 10:52:31.524 16373-16373/io.humon.app D/BluetoothManager: getConnectionState()
07-19 10:52:31.524 16373-16373/io.humon.app D/BluetoothManager: getConnectedDevices
07-19 10:52:31.528 16373-16373/io.humon.app D/BluetoothGatt: close()
07-19 10:52:31.528 16373-16373/io.humon.app D/BluetoothGatt: unregisterApp() - mClientIf=5
07-19 10:52:31.530 16373-16917/io.humon.app D/RxBle#Radio: FINISHED RxBleRadioOperationDisconnect(188299503)
07-19 10:52:33.547 16373-16856/io.humon.app W/ReactNativeJS: Possible Unhandled Promise Rejection (id: 0):
                                                             BleGattException{macAddress=00:22:D0:B0:C7:4C, status=133 (0x85 -> https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h), bleGattOperationType=BleGattOperation{description='CONNECTION_STATE'}}
                                                             createErrorFromErrorData@http://packager.dr-c6s.erhathaway.mobile-client.exp.direct:80/index.bundle?platform=android&dev=true&hot=false&minify=false:1964:24
                                                             http://packager.dr-c6s.erhathaway.mobile-client.exp.direct:80/index.bundle?platform=android&dev=true&hot=false&minify=false:1922:49
                                                             __invokeCallback@http://packager.dr-c6s.erhathaway.mobile-client.exp.direct:80/index.bundle?platform=android&dev=true&hot=false&minify=false:2237:21
                                                             http://packager.dr-c6s.erhathaway.mobile-client.exp.direct:80/index.bundle?platform=android&dev=true&hot=false&minify=false:2096:32
                                                             guard@http://packager.dr-c6s.erhathaway.mobile-client.exp.direct:80/index.bundle?platform=android&dev=true&hot=false&minify=false:2035:7
                                                             invokeCallbackAndReturnFlushedQueue@http://packager.dr-c6s.erhathaway.mobile-client.exp.direct:80/index.bundle?platform=android&dev=true&hot=false&minify=false:2095:12
                                                             invokeCallbackAndReturnFlushedQueue@[native code]
```

and 

```
07-19 10:57:00.953 16373-16856/io.humon.app I/ReactNativeJS: Found Polar
07-19 10:57:00.957 16373-16373/io.humon.app D/BluetoothAdapter: stopLeScan()
07-19 10:57:00.958 16373-16373/io.humon.app D/BluetoothAdapter: STATE_ON
07-19 10:57:00.958 16373-16373/io.humon.app D/BluetoothAdapter: STATE_ON
07-19 10:57:00.959 16373-16373/io.humon.app D/BluetoothAdapter: STATE_ON
07-19 10:57:00.959 16373-16373/io.humon.app D/BluetoothAdapter: STATE_ON
07-19 10:57:00.959 16373-16373/io.humon.app D/BluetoothLeScanner: Stop Scan
07-19 10:57:00.967 16373-16856/io.humon.app I/ReactNativeJS: is connected: false
07-19 10:57:00.971 16373-16856/io.humon.app I/ReactNativeJS: Connecting to Polar
07-19 10:57:00.977 16373-16857/io.humon.app D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(260546695)
[... cut ...]
                                                                   { _subscriber: 
                                                                      { _subscriptionsForType: 
                                                                         { appStateDidChange: 
                                                                            [ { subscriber: [Circular],
                                                                                emitter: 
```

The later might be more relevant to this ticket

erhathaway, 2017-07-19:
To clarify the later, the Expo XDE client shows:

```
10:56:57 AM
Scanning...
10:56:57 AM
Polar H7 B0C74C17
10:56:57 AM
Found Polar
10:56:57 AM
is connected: false
10:56:57 AM
Connecting to Polar
10:57:04 AM
Discovering Services and Characteristics
```

But the android studio logs are showing that the discovering finished. So maybe it is working, just not logging to the Expo console?

dariuszseweryn, 2017-07-19:
Hello @erhathaway — could you tell if your `Polar` is on the list of bonded devices on that Android Phone?

erhathaway, 2017-07-20:
@dariuszseweryn @Cierpliwy sorry it took long for my reply, I had to figure out how to do the native bridge to answer the bonded devices question.

The polar device is not on the list of bonded devices.

The code I ran was:
```
       Log.i("BluetoothStatus", "Fetching bonded devices");
       BluetoothAdapter mBluetoothAdapter = BluetoothAdapter.getDefaultAdapter();
       Set<BluetoothDevice> pairedDevices = mBluetoothAdapter.getBondedDevices();

       for(BluetoothDevice bt : pairedDevices) {
         Log.i("BluetoothStatus", bt.getName());
       }
```

erhathaway, 2017-07-20:
@dariuszseweryn @Cierpliwy it looks like we are now able to read values from the heartrate characteristic but we intermittently get the status 133 error still.

erhathaway, 2017-07-29:
Not an issue anymore. The repo update might have fixed this or maybe it was never an issue and I was just thrown off by the `console.log` statements not appearing in Expo... either way, closing the issue

closed 2017-07-29 by erhathaway