dotintent/react-native-ble-plx #108: ios - didModifyServices

brycejacobs, 2017-07-19:
Right now, we don't have the capability to receive notifications whenever services are invalidated on the native side.

[Reference](https://developer.apple.com/documentation/corebluetooth/cbperipheraldelegate/1518865-periphera)l

If that service is invalidated, and we end up trying to perform a procedure, we're going to run into problems. We also have no way of knowing when new services are added to the peripheral.

I guess the question is; how much effort is this to implement? I can see there being an easy way of implementing this, where either one or two events are emitted for the add / remove, but immediately I can see some book keeping headaches.

brycejacobs, 2017-08-02:
Hey guys, anything I can do to move this along?

Cierpliwy, 2017-08-04:
I think that RxBluetoothKit which is used internally by this library already provide needed function:
https://github.com/Polidea/RxBluetoothKit/blob/20d95560f069a242a755383121c2518c97592974/Source/Peripheral.swift#L539

We have internal "cache" or discovered services:
https://github.com/Polidea/react-native-ble-plx/blob/master/ios/BleClientManager/BleModule.swift#L29

I think we can monitor service changes after peripheral connection, update cache, rediscover any services and characteristics as needed and calculate three sets: added, removed and modified which will be send as events to JS side.

closed 2019-03-23 by stale[bot]