dotintent/react-native-ble-plx #109: Location needs to be enabled for BLE Scanning on Android 6.0

watadarkstar, 2017-07-20:
Just going to report this here for others.

Related to: https://stackoverflow.com/questions/33045581/location-needs-to-be-enabled-for-bluetooth-low-energy-scanning-on-android-6-0

Apparently the solution is: I solved this by setting targetSdkVersion to 22 in gradle file. You must declare ACCESS_COARSE_LOCATION in the manifest but, BLE scanning will work even if user denies this permission from App Settings.

I'm still testing if this works. I don't have my 6.0 android here on me right now.

watadarkstar, 2017-07-20:
@Cierpliwy What do you think of changing to `startDiscovery()`?

dariuszseweryn, 2017-10-11:
Using `BluetoothAdapter.startDiscovery()` does not provide any info about the `Advertisement Data` thus removing all BLE related advertisement part. This problem cannot be worked around unfortunately.

`BluetoothAdapter.startDiscovery()` could be used only for discovering the devices nearby and filtering by name or MAC address so the usability of this approach is very limited.

papayuca, 2018-04-18:
Hi,
Does this mean that it is not possible to use manager.startDeviceScan if the location services are disabled? 
:O

dariuszseweryn, 2018-04-18:
On `API>=23` — correct

closed 2019-03-23 by stale[bot]

BerndSchrooten, 2019-04-19:
It would be nice to mention this requirement in the docs.

omarbakeer, 2019-09-14:
@dariuszseweryn 

> On `API>=23` — correct

do you mean the sdk of this library or my application ?
I'm trying to downgrade the gradle compileSdkVersion but I've read that google won't accept my application on the play store if wasn't on at least using sdk version 26, so I tried to modify the react-native-ble-plx build.gradle from the source (node_modules) and it didn't change anything, still requires the location!

how should I ignore the Location to be turned on when searching for BT devices ?

dariuszseweryn, 2019-09-14:
> do you mean the sdk of this library or my application ?

API/OS version >= 23

> how should I ignore the Location to be turned on when searching for BT devices ?

You _should not_ ignore Location when searching for BLE devices

omarbakeer, 2019-09-20:
> You _should not_ ignore Location when searching for BLE devices

I really appreciate your support Dariusz, but my use case requires the minimum configuration, and I don't need the location or any of its related features (like the rssi). My question was if I targeted and used the **Sdk  22** would it be possible to use this library without enabling location? Thank you

dariuszseweryn, 2019-09-20:
> My question was if I targeted and used the Sdk 22 would it be possible to use this library without enabling location?

Yes — it would be possible. There are two caveats though:
1. [Google will not accept your app to the Play Store](https://developer.android.com/distribute/best-practices/develop/target-sdk)
2. I am not sure how if the system would return you scan advertisements if the user has disabled location services / GPS switch