dotintent/react-native-ble-plx #110: Error: more than one library with package name 'com.burnweb.rnpermissions'

nirmal25990, 2017-07-27:
New to react-native, trying to get example working. I am getting this error while trying to run example on android.

> FAILURE: Build failed with an exception.

>* What went wrong:
Execution failed for task ':app:processDebugResources'.
> Error: more than one library with package name 'com.burnweb.rnpermissions' 

Here is my package.json

```
{
  "name": "ReactBLEScanner",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "node node_modules/react-native/local-cli/cli.js start"
  },
  "dependencies": {
    "react": "16.0.0-alpha.6",
    "react-addons-pure-render-mixin": "15.6.0",
    "react-native": "0.44.1",
    "redux": "3.6.0",
    "react-redux": "5.0.5",
    "redux-logger": "3.0.6",
    "react-native-router-flux": "3.39.2",
    "immutable": "3.8.1",
    "redux-immutable": "4.0.0",
    "react-immutable-proptypes": "2.1.0",
    "react-native-button": "1.6.0",
    "react-native-android-permissions": "1.0.0",
    "buffer": "4.7.1",
    "react-native-ble-plx": "0.6.1"
  }
}
```
Are there any other examples on github or similar which would give me some
working examples to look at?

Thanks for any help if you can provide!

Cierpliwy, 2017-07-31:
Example app is not up to date unfortunately and I don't have time to fix this issue. I recommend to build new app and start by reading introduction in documentation.

nirmal25990, 2017-08-01:
Okay. No problem. Thanks for the reply.

closed 2019-03-23 by stale[bot]