dotintent/react-native-ble-plx #111: startDeviceScan not working with array of strings containing UUID s of Service

jchirschy, 2017-07-28:
Hi @Cierpliwy,

When I use startDeviceScan with one of the following, everything works fine :
1. this.manager.startDeviceScan with UUIDs = `null` 
1. this.manager.startDeviceScan with UUIDs = `[]`
1. this.manager.startDeviceScan with UUIDs = `['valid-services-uuids']` like for Nordic `['6e400001-b5a3-f393-e0a9-e50e24dcca9e']`

But when I use startDeviceScan with an array of more than one valid UUIDs, app is keeping on scanning. For example with `[['6e400001-b5a3-f393-e0a9-e50e24dcca9e', '713d0000-503e-4c75-ba94-3148f18d941e']` (Nordic and redbear).

Did I misuse this method or something goes wrong ?
I use it on an Android device. Not tested with IOS.

Thanks for your help
Sincerily
JeanClaude

dariuszseweryn, 2017-07-28:
Hello @jchirschy 
As far as I know both UUIDs need to be present in the device's advertisement to match it with the filter. Do you have a device that advertises both UUIDs?

jchirschy, 2017-07-29:
Hello @dariuszseweryn,

Thanks for this explanation.
I thought it was a way to filter out devices by services they are advertising.
Is there a way to do that properly using this library ? I mean, what is the best way to list compatible devices with some services other by invoking this.manager.startDeviceScan for each services ?

Thanks

Cierpliwy, 2017-07-31:
If you want to filter your device by specific UUIDs, you have to be sure that every UUID which you are setting in `this.manager.startDeviceScan` has to be available in advertisement data of your device. If that doesn't work then there is a bug to be fixed.

jchirschy, 2017-07-31:
Thanks you all,

I had misunderstood how this method works. I thought the array was used as an OR condition.
It is actually an AND condition.

What I'm looking for is listing UART compatible devices, that means, each compatible device may have just only one compatible service advertised. Compatible devices can use different services for accomplishing the same thing.

I suppose I have to do another way. Maybe scanning without specifying any UUIDs, and then filtering out only compatible services.

closed 2017-07-31 by jchirschy

dariuszseweryn, 2017-07-31:
@jchirschy Yes — filtering on your own is a valid approach
@Cierpliwy Maybe it could be worth to put an emphasis on the `AND` aspect for the `UUID`s list so people will not get confused

askielboe, 2018-01-25:
What is the reasoning behind the decision to only return devices that include all UUIDs?

This goes against the default behaviour of CoreBluetooth, which always returns all devices that have any of the passed UUIDs.

dariuszseweryn, 2018-01-25:
On Android the default behaviour is opposite. You have to pick one when implementing a shared library.

Best Regards

askielboe, 2018-01-25:
Thanks for the reply!

Not sure if this is a historical thing on Android, but the new version of `RxAndroidBle` has deprecated the scanning method used by `react-native-ble-plx` which performs the filtering:

```
@Deprecated
public abstract Observable<RxBleScanResult> scanBleDevices(@Nullable UUID... filterServiceUUIDs)
```

The new scan method supports custom scan filters, allowing you to choose exactly how you'd like to filter the scanned devices:

```
public abstract Observable<ScanResult> scanBleDevices(ScanSettings scanSettings, ScanFilter... scanFilters);
```

Seems that `RxBluetoothKit` has also removed filtering in 5.0 and now only support running a single scanner:

https://github.com/Polidea/RxBluetoothKit/commit/9a3eb7669d4b38d20294d97d70257a9fa6624a9f

Not sure if this is a conscious decision, but it would definitely make scanning for devices with different services a lot easier if one could use a single scanner observable with custom filtering.

I realize this may require some changes in `react-native-ble-plx` to work properly, and I'm not asking that you do anything about it, simply putting it here in case anyone's interested. :)

Cheers!