dotintent/react-native-ble-plx #112: Base64

kompost, 2017-07-31:
Hello, I have trouble understanding the characteristic value output. I'm working on genuno101 board, for testing purposes I simply increment an index to a 100 and update bluetooth with the value accordingly. As I understand the value is encoded in Base64 and needs to be decoded. 
fx the number 4 encoded in base64 is _BA==_, but what I get is _BAAAAA==_ which is not exactly 4. 
Do I need some library to decode or im I not understanding this at all? :D

Cierpliwy, 2017-07-31:
Everything depends on your internal characteristic's value representation. From what you just wrote I assume that it is 4 byte integer encoded as little endian. So to increment value you could do:
```
let value = Buffer.from("BAAAAA==", "base64")
let number = value.readInt32LE()
let newNumber = number + 1
let newValue = Buffer.alloc(4)
newValue.writeInt32LE(newNumber)
let base64 = newValue.toString('base64')
```

kompost, 2017-07-31:
Thank you, worked like a charm :)

closed 2017-07-31 by kompost