dotintent/react-native-ble-plx #113: linker commander failed with exit code 1

dubzelecta, 2017-07-31:
hello , getting this error when trying to build on xcode -

framework not found  bleclientmanager for arm64

RN 0.44
xcode 8.3.3

ShaunSpringer, 2017-08-17:
+1 Getting the same issue here

RN 0.47.1
XCode 9 Beta 5 (9M202q)

`{
  "name": "twistapp",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "node node_modules/react-native/local-cli/cli.js start",
    "test": "jest"
  },
  "dependencies": {
    "babel-preset-react-native": "2.0.0",
    "binary-parser": "^1.1.5",
    "buffer": "^5.0.7",
    "immutable": "^3.8.1",
    "react": "16.0.0-alpha.12",
    "react-native": "0.47.1",
    "react-native-ble-plx": "^0.6.1",
    "react-native-config": "^0.6.0",
    "react-native-vector-icons": "^4.3.0",
    "react-navigation": "^1.0.0-beta.11",
    "react-redux": "^5.0.6",
    "redux": "^3.7.2",
    "redux-logger": "^3.0.6",
    "redux-saga": "^0.15.6",
    "reselect": "^3.0.1"
  },
  "devDependencies": {
    "babel-jest": "20.0.3",
    "jest": "20.0.4",
    "react-test-renderer": "16.0.0-alpha.12"
  },
  "jest": {
    "preset": "react-native"
  }
}
`

mikewrather, 2017-09-26:
I had an issue with automatically linking with `react-native link`.  Follow the instructions for manually setting up the library here [Configuration and Settings](https://github.com/Polidea/react-native-ble-plx#configuration--installation)

Cierpliwy, 2018-04-09:
Closing due to inactivity.

closed 2018-04-09 by Cierpliwy