dotintent/react-native-ble-plx #115: Datas of notification are lost

adyanced, 2017-08-01:
My ble device samples and stores temperatures and it sends these datas to android app when notification is opened. Every 20 bytes per frame has a header indicating its order , totally about 160 frames. The ble device works well.

RN code as below:
 
componentDidMount() {
    transactionId = 'monitor';

    let deviceId = this.props.navigation.state.params.key
    this.setState({deviceId: deviceId})
//    setTimeout(() => {
      _bleManager.monitorCharacteristicForDevice(deviceId, gatt.UUID_HUM_SERVICE, gatt.UUID_HUM_DATA, (error, characteristic) => {
        if (error) {
          if (error.message === "Cancelled")  return
          ToastAndroid.show(error.message, ToastAndroid.SHORT);
          return
        }

//        let data = db.writeRecord(new Buffer(characteristic.value, "base64"))
        let data = new Buffer(characteristic.value, "base64")
        console.log(data)
        this.setState({dataReceived: this.state.dataReceived + data.length})
      }, transactionId)
//    }, 5000);        

	};

The issue is app always lost the first three frames of notification data, it receives only 3140 bytes. Same problem for less datas. It does not work to delay notification adding 'setTimeout'.

Another question is I described in this thread [https://github.com/Polidea/react-native-ble-plx/issues/68](url) . Is 'cancelTransaction' must called before 'cancelDeviceConnect'?

adyanced, 2017-08-01:
Not always first 3 frames, may be more or less are lost. But always the first several frames are lost

mswokk, 2017-12-08:
Perhaps duplicate of #152
Please try it.

Cierpliwy, 2018-01-05:
This should be fixed with version 0.8.0. Please reopen if there are any problems.

closed 2018-01-05 by Cierpliwy