dotintent/react-native-ble-plx #119: Unable to Install the library

cchaitu05, 2017-08-01:
Hi All,

I am trying to add this library to my project. I have added the dependency and when try to install i am facing the below issue. Can someone help?

npm ERR! Error while executing:
npm ERR! C:\Program Files\Git\cmd\git.EXE ls-remote -h -t ssh://git@github.com/Polidea/react-native-ble-plx-master.git
npm ERR!
npm ERR! Host key verification failed.
npm ERR! fatal: Could not read from remote repository.
npm ERR!
npm ERR! Please make sure you have the correct access rights
npm ERR! and the repository exists.
npm ERR!
npm ERR! exited with error code: 128

deepupathak, 2017-10-05:
try to remove node_modules and again install this plugin.

Cierpliwy, 2018-04-09:
Closing due to inactivity.

closed 2018-04-09 by Cierpliwy