dotintent/react-native-ble-plx #122: Core missing from react-native Libraries

cinvoke, 2017-08-05:
When i npm install the example ble scanner, its missing the Core library in node_modules.  When i start a new react-native project, it contains this.  
1. Why is Core missing?
2. im trying to fix a syntax error i get when running this example on my device.  
https://stackoverflow.com/a/41059867/2792358

Thx.

cinvoke, 2017-08-05:
react-native-cli: 2.0.1
react-native: 0.30.0

cinvoke, 2017-08-05:
seems like the package.json is already trying to fix this issue:
`{
  "name": "ReactBLEScanner",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "node node_modules/react-native/local-cli/cli.js start"
  },
  "dependencies": {
    "buffer": "4.7.1",
    "immutable": "3.8.1",
    "react": "15.2.1",
    "react-addons-pure-render-mixin": "15.2.1",
    "react-immutable-proptypes": "2.0.0",
    "react-native": "^0.30.0",
    "react-native-android-permissions": "0.0.6",
    "react-native-ble-plx": "Polidea/react-native-ble-plx",
    "react-native-button": "1.6.0",
    "react-native-router-flux": "3.32.0",
    "react-redux": "4.4.5",
    "react-update": "^0.4.4",
    "redux": "3.5.2",
    "redux-immutable": "3.0.6",
    "redux-logger": "2.6.1"
  },
  "devDependencies": {
    "babel-plugin-transform-es2015-block-scoping": "^6.24.1"
  }
}
`

Cierpliwy, 2018-04-09:
Old example project is deprecated. Closing.

closed 2018-04-09 by Cierpliwy