dotintent/react-native-ble-plx #124: Every time I yarn add / remove other framework plx would be build scheme over again

CrazyWr, 2017-08-10:
[image]

CrazyWr, 2017-08-10:
and I remove other framework the dictionary `react-native-ble-plx/ios/BleClientManager/Carthage` would be deleted

wkoutre, 2017-09-22:
Ever find a solution to this?

mattermoran, 2017-11-07:
Still nothing?

derrickbeining, 2018-06-01:
@Cierpliwy, this still appears to be an issue. Every time I npm/yarn install something in my project, the bluetooth package rebuilds and takes forever. Do you have any idea what is causing that and how it could be fixed? If you can give me a clue, I'd be happy to try my hand at fixing and make a PR.

derrickbeining, 2018-06-01:
I guess using the `--ignore-scripts` flag could also be a fine work around for now too

closed 2019-03-23 by stale[bot]