dotintent/react-native-ble-plx #125: Can not run 'long live' in the background

adyanced, 2017-08-11:
Hi,
I found it can only run 25minutes if the screen is off and running in the background. After that scanning is stopped but the app is not killed. 
Is it possible to scan ble in the background as long as phone is powered?

grigored, 2017-08-13:
Can you post the code you used to configure background mode to make it work even for the first 25 minutes? I couldn't make it work in background at all, and posted [this question](https://github.com/Polidea/react-native-ble-plx/issues/127). I'm also interested in getting a long lived background mode.

samthui, 2018-01-09:
In which case(s) you need to keep app scanning in background?

rgstephens, 2018-01-12:
As with @adyanced, I need this to scan 24/7.  In my case, background scanning is the primary use of my app to monitor for the presence of beacons.

closed 2019-03-23 by stale[bot]

roysG, 2019-08-28:
Any updatee?