dotintent/react-native-ble-plx #126: The following binaries use incompatible versions of Swift

grigored, 2017-08-12:
I followed the install instructions, the app runs on ios simulator (though state is first Unknown and then Unsupported), but when I try to install the app on a real device, i get:  
```
CopySwiftLibs
error: The following binaries use incompatible versions of Swift:
The following binaries use incompatible versions of Swift:
/Users/grigore/Library/Developer/Xcode/DerivedData/myProject-awzikblhwjhwbgciqztjkrsyeruy/Build/Products/Debug-iphoneos/instacarFront.app/Frameworks/BleClientManager.framework/BleClientManager
/Users/grigore/Library/Developer/Xcode/DerivedData/myProject-awzikblhwjhwbgciqztjkrsyeruy/Build/Products/Debug-iphoneos/myProject.app/myProject```
Any ideas on what I'm doing wrong?

grigored, 2017-08-12:
I had both xcode 8.3.3 and xcode 9.0 beta installed, and was trying to build with xcode 9 beta. Carthage was using swift 3.0 to build BleClientManager framework, but xcode 9.0 beta was using swift 3.2. I just switched to xcode 8 and it worked.

closed 2017-08-12 by grigored

jiazil, 2017-10-03:
it worked ! but there isn't the final solution I think，because we will keep the xcode 8.3.3 forever.hope someone can resolute it.

ayu250, 2017-10-10:
rebuild the framework with xcode9.0 and fixed ...., xcode 8.3.3 build framework with swift 3.1 ,and xcode 9  build framework with swift3.2 , they were incompatible ,,,, swift 4.0 will be fine, so upgrade your code to swift 4.0 will fix it too .

jay0lu, 2018-04-04:
Try use `watchman watch-del-all && rm -rf $TMPDIR/react-* && rm -rf node_modules/ && npm cache clean && npm install ` to clean the cash. It works for me on xcode9.2.