dotintent/react-native-ble-plx #127: can't make background mode work

grigored, 2017-08-13:
My app is supposed to send a message to a specific BLE device when that device is in range. Ideally this should work even when the app is closed/in background mode. I used the simplest `restoreStateFunction` I could think of: it just makes a `fetch` request to an endpoint on my server. However, this is never called once the app goes to background. I created a [sample repo](https://github.com/grigored/testble) to quickly replicate the problem, and here's the relevant code. Tested on android/ios, debug/release and nothing works. Any idea about what I'm doing wrong? 

Also, I only want the app to be turned on only if a specific device is in range, and I see this is configured only in `startDeviceScan`. Will the background mode take this into account, or the app will be turned on anytime a new ble device is in range?  

```
import React from 'react';
import {BleManager} from "react-native-ble-plx";
import {PermissionsAndroid, Platform} from 'react-native';

export class BleComponent extends React.PureComponent {
    constructor() {
        super();
        if (Platform.OS === 'android') {
            PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION)
                .then(granted => {
                    if (granted !== 'granted') {
                        console.log('BLUETOOTH denied')
                    }
                    else {
                        this.initializeManager();
                    }
                })
        } else {
            this.initializeManager();
        }

    }

    initializeManager() {
        this.manager = new BleManager({
            restoreStateIdentifier: 'testBleBackgroundMode',
            restoreStateFunction: bleRestoredState => {
                fetch('http://restoreStateFunctionWasCalled.com')
            }
        });
        const subscription = this.manager.onStateChange((state) => {
            if (state === 'PoweredOn') {
                this.scanAndConnect();
            }
        }, true);
    }

    scanAndConnect() {
        console.log(('scan and connect'));
        this.manager.startDeviceScan(["0000fefb-0000-1000-8000-00805f9b34fb"], null, (error, device) => {
            if (error) {
                console.log(("BLE error"));
                return
            }
            console.log(device);
        });
    }

    render() {
        return null;
    }
}
```

adyanced, 2017-08-14:
Hi, there is no any special codes in my app. Can your codes scan any devices cause I think maybe

grigored, 2017-08-14:
Thanks for your help. I tested that as well, doesn't help. Scanning works as expected while the app is in foreground (on Android also when in background if i don't close it - but this has nothing to do with the BLE background mode, as everything continues to work, not only the ble scanning).

adyanced, 2017-08-16:
@grigored I think what you want is not a job of react-native-ble-plx. I only need ble scanning in the background. But after a half day's test, I'm considering to turn to another ble project because of three reasons:
1. react-native-ble-plx always lose first several notification data chunks, but another one not. I reported this issue which makes it unusable for me.
2. react-native-ble-plx can only scan 25 minutes in the background, but that one can scan so long as app is live.
3. react-native-ble-plx gives base64 data to user, but that one transmits binary array. It means a little bit more faster cause I have about 3000 notification bytes after connected.
Anyway react-native-ble-plx is still a awesome project.

Cierpliwy, 2017-08-16:
@adyanced Well mostly it is a job of this library :)

On iOS background mode is tricky. Just adding special permission in XCode project allows you to do background work. There is one special case when system is taking control of your BLE connection and kills your application. Restoration function is used to allow to restore your BLE state after application relaunch (when user is going to foreground after long period in background mode). Personally I only tested if connection is kept in this situation. You can simulate this behaviour by sending SIGKILL signal to your iOS process from lldb debugger.

1). That's probably a bug which should be fixed.
2). I didn't test scanning functionality. Would love contributions.
3). react-native-ble-plx gives base64 to user because it is the only way to send binary data between native code and JS. There were attempts to add support for ArrayBuffer, but I guess it wasn't implemented in RN.

DVassilev, 2017-09-11:
Have there been any updates on getting background mode to work?

konradrodzik, 2017-09-11:
Did you check latest v0.6.1 ? Background mode was merged in v0.6.0 and there were some fixes in v0.6.1
https://github.com/Polidea/react-native-ble-plx/releases

lastmjs, 2018-05-29:
@grigored and @adyanced Were either of you able to get background scanning to work with Android or iOS? I'm about to start implementing these features and I want to know if this library will be capable. So far I'm finding various issues that is making this library difficult to deal with.

harshjoshi096, 2018-08-31:
Did any one get the solution for connecting and scanning bluetooth device in background for iOS app?
@grigored @adyanced @Cierpliwy @konradrodzik

EskelCz, 2019-02-09:
> @grigored I think what you want is not a job of react-native-ble-plx. I only need ble scanning in the background. But after a half day's test, I'm considering to turn to another ble project because [...]

What other library are you comparing this to? Also have you solved the background scanning? Thanks

closed 2019-04-17 by stale[bot]

good-lly, 2019-04-26:
2019 - is there is any solution for that now?

I setup background capabilities in xcode/iOS as well as setup `restoreStateFunction` but background mode is not able to handle BLE events when app is killed or in background.

Did you try [react-native-background-task](https://github.com/jamesisaac/react-native-background-task) is it valid workaround?

EskelCz, 2019-04-26:
There is no easy solution, until apple changes its policies.
The only option is reverse engineering, as described here:
https://crownstone.rocks/2018/06/27/ios-advertisements-in-the-background

farzd, 2019-08-02:
Anyone tried updating their plist?

https://stackoverflow.com/questions/49802451/react-native-native-module-possibilities-for-background-task-on-ios

julioxavierr, 2019-09-25:
Any workaround on this?

7772, 2019-11-05:
I have trouble on scanning device on background, but solved it.

1. make sure turn on background mode on Xcode.
2. make sure that properly pass service uuids parameter in `startDeviceScan `

I noticed what I should do after reading on [this](https://stackoverflow.com/a/20460113/7770508)
```
The problem is the difference in scanning in foreground and background. 
When you are scanning for devices in the foreground you can scan for anything. 
In the background you must specify the actual service UUID you are scanning for. 
Ok, this isn't actually a problem as you know the UUID you are looking for.
```

priyanka-Sadh, 2020-06-04:
@7772 It's great to hear form you that it's worked.
Even to put correct serviceUUID it doesn't work. Can you please make a proper documentation on background mode os scan and can share, I think it will be help ful for lot of people. That will be grateful though :) 
Or someone can say it never work :(

azuxx, 2020-07-02:
> 1. make sure turn on background mode on Xcode.

@7772 which one? I have this currently:

[image]