dotintent/react-native-ble-plx #129: Reading the MTU?

kirakik, 2017-08-16:
Is there currently a way to detect the current MTU? In iOS `CBPeripheral`s have a function called `maximumWriteValueLength` which can be used for this purpose, but I could not find anything related to that in the documentation.

Cierpliwy, 2017-10-24:
Thanks again for your work. Version 0.6.4 added `mtu` property to `Device` object which allows you to read it after device is connected with `requestMTU` option specified or by calling `requestMTUForDevice` function.  On iOS requesting MTU is noop operation as iPhone is doing it automatically (however after a while when device is connected it should be available and in most cases it is 158 bytes). On Android make sure to call one of above functions.

juhanilamma, 2017-10-27:
To me it looks like:

On 0.6.4 when cancelDeviceConnection() is called on BleModule.java and promise is being resolved successfully:
promise.resolve is called by argument "new Device(device, null).toJSObject(null)"
=> So the given new device object is initialized with null connection.

New device is then called "toJSObject()", which executes line:
"result.putInt(Metadata.MTU, connection.getMtu());"

This causes NPE as the connection is null and connection.getMtu() is called.

Could you please check this if I read the code correctly.

Cierpliwy, 2017-10-28:
You are right. Will fix with new version.

pstanton, 2017-11-23:
I'm not currently testing on iOS, just android, but can you explain if the following is correct?

```
device.connect({requestMTU:512}).then(deviceC => {
	console.log(`connected ${deviceC.id} ${deviceC.name} mtu:${deviceC.mtu}`);
});
```

hypothetically, if the peripheral only supports 300 and i'm on an iOS device, the mtu in the log message should (usually) be 158. if the peripheral only supports 100, the mtu in the message should be 100.

ie if i request the max (512) does the api sort out the min of my request, the device and the peripheral?

Cierpliwy, 2017-11-23:
On iOS you can't request MTU and JS call won't do anything. On Android we are calling https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#requestMtu(int). Still documentation is not clear here. I think you have to check in practice how device behaves.

Cierpliwy, 2017-11-30:
Closing this issue as changing MTU was implemented and NPE fixed in 0.6.5. Please open another issue for further discussion or just ask us on Gitter.

closed 2017-11-30 by Cierpliwy