dotintent/react-native-ble-plx #131: I can't write more than 16 bytes

guerjon, 2017-08-18:
I'm getting this error Error: BleGattCharacteristicException status = 0xd, this error happens because I try to write a value between 16 and 20 bytes, I know that you can't write more than 20 bytes but why I can't write 17,18,19 or 20 bytes? 

I directly change your method writeCharacteristicWithValue on the BleModule and initialized the 
final byte[] value = {
                0x1,0x1,0x1,0x1,0x1,
                0x1,0x1,0x1,0x1,0x1,
                0x1,0x1,0x1,0x1,0x1,
                0x1}
for testing and it works, but if you add another byte throw the error so you can only write 16 bytes no more, if you try you get this error.
Any idea about why it is happening?

dariuszseweryn, 2017-10-11:
Apparently the characteristic you try to write does not allow for writing more than 16 bytes. `0x0d` -> https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h#41

guerjon, 2017-10-11:
Nop, I'm able to write 20 bytes in the same characteristic with just android.

dariuszseweryn, 2017-10-11:
Tested using the same device?

guerjon, 2017-10-11:
I've been testing on distinct android phones and same result, or you mean on the bridge?

dariuszseweryn, 2017-10-11:
What would be valuable is to:

1. Run your `RN` app that is trying to write 20 bytes on your characteristic
2. Run a native android app that is trying to write 20 bytes on your characteristic using `RxAndroidBle` library
3. Run a native android app that is trying to write 20 bytes on your characteristic using the vanilla API

I would expect all three approaches to return the same result if executed on the same phone/peripheral pair.

guerjon, 2017-10-17:
i wrote correctly with the vanilla api but, in the same ble device, but i have not been able to write with your library, i just can write 16 bytes correctly

guerjon, 2017-10-17:
this is the complete error message 
{ [Error: BleGattCharacteristicException{macAddress=EB:16:3A:07:E2:44, status=13 (0x0d -> https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h), bleGattOperationType=BleGattOperation{description='CHARACTERISTIC_WRITE'}}]

dariuszseweryn, 2017-10-18:
Could you share the complete code snippets for both using the `react-native-ble-plx` and the `vanilla` API so we could compare them?

lnanhkhoa, 2018-01-02:
I'm getting the same error. Anybody help me?

lnanhkhoa, 2018-01-02:
I solved it. If device need n bytes, I should writes n bytes on ReactNative App. 
I've been testing in 20 bytes and i't's OK

dariuszseweryn, 2018-01-03:
I do not get it. Could you elaborate more on what was wrong in your original approach?

lnanhkhoa, 2018-01-03:
The `react-native-ble-plx` doesn't wrong. I'm sure of that.
This error maybe appear from device. With (status = 13) said that your device doesn't allow you write many bytes).
You can see [here](https://imgur.com/a/Q85Ip).
So, you need fix it or write obligated bytes (normally only 1-byte).
I use CC1350 launchpad to develop BLE application.
**In RN app**: You create a Array that includes n bytes you want. This array need convert base64-format and write like Sniffator example.
Good luck !!

dariuszseweryn, 2018-01-03:
So it seems that the problem was an incorrect handling of the payload data. Remember to always use `Base64` encoding and not raw byte arrays.

closed 2018-01-03 by dariuszseweryn