dotintent/react-native-ble-plx #132: App crashes. dyld: Library not loaded: Reason: no suitable image found.  Did find: BleClientManager

OlegPanfyorov, 2017-08-25:
Hi guys, I don't know what to do, can you help me please? My app runs but crashes immediately. I've added package automatically through npm, also checked all manual steps (same result). Here is the log:

dyld: Library not loaded: @rpath/BleClientManager.framework/BleClientManager
  Referenced from: /var/containers/Bundle/Application/C3B47D9A-F545-4625-B49A-D000D9AA24D2/MyAppName.app/MyAppName
  Reason: no suitable image found.  Did find:
	/private/var/containers/Bundle/Application/C3B47D9A-F545-4625-B49A-D000D9AA24D2/MyAppName.app/Frameworks/BleClientManager.framework/BleClientManager: code signing blocked mmap() of '/private/var/containers/Bundle/Application/C3B47D9A-F545-4625-B49A-D000D9AA24D2/MyAppName.app/Frameworks/BleClientManager.framework/BleClientManager'

OlegPanfyorov, 2017-08-28:
**Fixed**
The problem was with Apple Application Integration Certification Authority certificate and signing Carthage dependencies. Removed certificate from Keychain, added new one from here https://developer.apple.com/support/certificates/expiration/ and now everything work perfect

closed 2017-08-28 by OlegPanfyorov

kevando, 2018-06-14:
How did you realize this was the issue? I tried this, still getingg this error.