dotintent/react-native-ble-plx #133: NullPointerExceptioncom.polidea.reactnativeble.wrapper.Device in getServiceByUUID

soyangel, 2017-08-30:
Hi,

I'm getting the following error when doing a "writeCharacteristicWithoutResponseForDevice".

Please let me know what aditional data you need to debug this.

---

java.lang.NullPointerException: Attempt to invoke interface method 'java.util.Iterator java.util.List.iterator()' on a null object reference
    at com.polidea.reactnativeble.wrapper.Device.getServiceByUUID(Device.java:53)
    at com.polidea.reactnativeble.BleModule.getCharacteristic(BleModule.java:892)
    at com.polidea.reactnativeble.BleModule.writeCharacteristicForDevice(BleModule.java:547)
    at java.lang.reflect.Method.invoke(Method.java)
    at com.facebook.react.bridge.BaseJavaModule$JavaMethod.invoke(BaseJavaModule.java:368)
    at com.facebook.react.cxxbridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:138)
    at com.facebook.react.bridge.queue.NativeRunnable.run(NativeRunnable.java)
    at android.os.Handler.handleCallback(Handler.java:751)
    at android.os.Handler.dispatchMessage(Handler.java:95)
    at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:31)
    at android.os.Looper.loop(Looper.java:159)
    at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:208)
    at java.lang.Thread.run(Thread.java:761)

Cierpliwy, 2017-08-30:
Thanks for your report. I know where the bug is located in native code.

You can temporarily get rid of it making sure that your device discovered all services and characteristics before performing any actions on it:
https://polidea.github.io/react-native-ble-plx/#Device#discoverAllServicesAndCharacteristics

Cierpliwy, 2017-11-30:
It was fixed in version 0.6.3.

closed 2017-11-30 by Cierpliwy