dotintent/react-native-ble-plx #134: PPCP

kompost, 2017-08-30:
Hey

I need to set my own connection interval, but it seems there is no such option. This is from the documentation, where I assume the PPCP is set:
"ConnectionOptions
Connection specific options to be passed before connection happen. [Not used]"
How can I get around this issue?

Cierpliwy, 2017-08-31:
I have never seen iOS or Android API which would allow you to set PPCP of BLE connection. Can you send me a link to documentation which describes that?

kompost, 2017-09-01:
Hmm, 
Long story short I need to update a characteristic at 100hz, or as close as possible. Basically if I don't set any connection parameters the bluetooth crashes. There is an app for bluetooth inspection called nRF Connect and in that app it is possible to set PPCP. See the attached picture.
[image]
If it is set on high my app runs without crashing and sending all the data. I assume thats PPCP that is set there or am I missunderstanding something?

dariuszseweryn, 2017-10-28:
On Android the function for this functionality is named `BluetoothGatt.requestConnectionPriority(int)`. It is available on the level of `RxAndroidBle` underneath starting from API 21.

I am not sure if iOS does allow to change these settings which may be a problem to implement a function on the `react-native-ble-plx` level that would work the same on both platforms. Moreover iOS does not allow for Connection Intervals shorter than 30 ms if I am not mistaken.

An additional problem on Android is a bit of inconsistency of what exact parameters you would get by requesting a specific priority on different OS versions.

The best idea would be to alter the firmware of your peripheral to manage the connection parameters as from the peripheral side one would get most control over the procedure.

Best Regards

JBaczuk, 2020-05-19:
@dariuszseweryn It appears this might be the API on iOS to change the connection interval:

https://developer.apple.com/documentation/corebluetooth/cbperipheralmanager/1393277-setdesiredconnectionlatency

If so, how would one go about implementing this in the library?

dariuszseweryn, 2020-05-21:
This actually looks like it could be doable now. I will mark this issue as a feature request.

lifekaizen, 2020-07-02:
> @dariuszseweryn It appears this might be the API on iOS to change the connection interval:
> https://developer.apple.com/documentation/corebluetooth/cbperipheralmanager/1393277-setdesiredconnectionlatency

Note that this method says: `When you call this method ... connection latency changes aren’t guaranteed. `

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska

lcrodrigues, 2024-11-11:
@abursztynowska is there any way to change the connection interval in the new version of `react-native-ble-plx`?