dotintent/react-native-ble-plx #135: Build failed with Xcode 9 beta 6

jamescoletti, 2017-09-01:
Is this project compatible with Xcode 9 beta yet? At the very least, the build seems to fail in the RxBluetooth iOS dependency, which is set at 3.0.12, and it appears there have been some Xcode 9 beta related fixes in 3.0.14.

jamescoletti, 2017-09-07:
FYI - I was able to get everything working by:

1. Bumping the RxBluetooth dependency version from 3.0.12 to 3.0.14
2. Changing the `#import` statements per PR #136  

Using `react-native@0.48.1` and `react-native-ble-plx@0.6.1`

mmerickel, 2017-09-20:
I can confirm this fixed my issues for now as well.

SchwarzerSchnee, 2017-09-21:
Doesn't work for me because I'm getting the files to be changed only with npm install, but they'll be deleted right again upon failure.

jamescoletti, 2017-09-21:
@SchwarzerSchnee you could fork the repo, make the changes, and then `npm install` your forked version. Or for a quick-and-dirty solution after install failure, you could probably update the versions in the Cartfile in `ios/BleClientManager`, delete the `Carthage` folder there, and run `bash build_ios_frameworks.sh`.

SchwarzerSchnee, 2017-09-25:
Thank you, that works for now but still it's kinda dirty to have a local dependency that shouldn't be there. Here's hoping the issue gets fixed entirely on the lib's end eventually, since Xcode 9 has been officially released now.

Cierpliwy, 2017-09-25:
Yesterday I downloaded XCode 9, so hopefully I'll release fixes and new features this week with 0.7.0 version. Sorry for delay.

fisehara, 2017-09-26:
Confirm changes to :
Carfile: Github "Polidea/RxBluetoothKit" "3.0.14"
Cartfile.resolved: github "Polidea/RxBluetoothKit" "3.0.14"

solved my issue as well

Cierpliwy, 2017-09-28:
This issue should be resolved with version 0.6.2.

jamescoletti, 2017-09-28:
Thanks @Cierpliwy! I can confirm 0.6.2 is building successfully in Xcode 9.

closed 2017-11-30 by Cierpliwy