dotintent/react-native-ble-plx #141: xcode 9

jiazil, 2017-09-20:
npm install ,the error:
> react-native-ble-plx@0.6.1 postinstall /node_modules/react-native-ble-plx
> bash build_ios_frameworks.sh

*** Checking out RxBluetoothKit at "3.0.12"
*** Checking out RxSwift at "3.4.1"
*** xcodebuild output can be found in /var/folders/dr/lcfvwdvx4lqgvbsgtf6kkkqr0000gn/T/carthage-xcodebuild.ERCMxg.log
*** Building scheme "RxBlocking-iOS" in Rx.xcworkspace
*** Building scheme "RxCocoa-iOS" in Rx.xcworkspace
*** Building scheme "RxTests-iOS" in Rx.xcworkspace
*** Building scheme "RxSwift-iOS" in Rx.xcworkspace
*** Building scheme "RxBluetoothKit iOS" in RxBluetoothKit.xcodeproj
Build Failed
        Task failed with exit code 65:
       /node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxBluetoothKit/RxBluetoothKit.xcodeproj -scheme RxBluetoothKit\ iOS -configuration Release -derivedDataPath /Library/Caches/org.carthage.CarthageKit/DerivedData/9.0_9A235/RxBluetoothKit/3.0.12 -sdk iphoneos ONLY_ACTIVE_ARCH=NO BITCODE_GENERATION_MODE=bitcode CODE_SIGNING_REQUIRED=NO CODE_SIGN_IDENTITY= CARTHAGE=YES build (launched in /node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Checkouts/RxBluetoothKit)

This usually indicates that project itself failed to compile. Please check the xcodebuild log for more details: /var/folders/dr/lcfvwdvx4lqgvbsgtf6kkkqr0000gn/T/carthage-xcodebuild.ERCMxg.log
npm WARN react-native-pdf-view@0.4.0 requires a peer of @types/react@* but none is installed. You must install peer dependencies yourself.
npm WARN react-native-pdf-view@0.4.0 requires a peer of @types/react-native@* but none is installed. You must install peer dependencies yourself.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! react-native-ble-plx@0.6.1 postinstall: `bash build_ios_frameworks.sh`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the react-native-ble-plx@0.6.1 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

zbtang, 2017-09-21:
Same error here

cmeredith, 2017-09-21:
Same error.

wkoutre, 2017-09-22:
Dupe of https://github.com/Polidea/react-native-ble-plx/issues/135

Cierpliwy, 2017-09-28:
This issue should be resolved with version 0.6.2.

askielboe, 2017-09-28:
I have the same issue with version 0.6.2 and Xcode 9.

Cierpliwy, 2017-09-28:
I can't reproduce it on XCode 9. Can you paste your error?

askielboe, 2017-09-29:
Sorry, didn't have time to debug yesterday.

My build fails when it tries to run swiftlint.

`brew uninstall swiftlint` fixes the problem for me.

jiazil, 2017-10-03:
if i upgrade to 0.6.2，there is other wrong: error: The following binaries use incompatible versions of Swift

Cierpliwy, 2017-10-03:
Make sure that all frameworks which you use in your project were compiled with the same compiler version. You can force rebuilding frameworks use by this library by reinstalling react-native-ble-plx.

jiazil, 2017-10-03:
the error detail:
CopySwiftLibs /Users/yifeilin/Library/Developer/Xcode/DerivedData/lgapp-etnuzaqrmdesiefbjjihipkqilio/Build/Products/Release-iphonesimulator/leisure.app
    cd /Users/yifeilin/project/core/leisure/ios
    export CODESIGN_ALLOCATE=/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/codesign_allocate
    export DEVELOPER_DIR=/Applications/Xcode.app/Contents/Developer
    export PATH="/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/usr/bin:/Applications/Xcode.app/Contents/Developer/usr/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin"
    export SDKROOT=/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator11.0.sdk
    builtin-swiftStdLibTool --copy --verbose --sign - --scan-executable /Users/yifeilin/Library/Developer/Xcode/DerivedData/lgapp-etnuzaqrmdesiefbjjihipkqilio/Build/Products/Release-iphonesimulator/leisure.app/leisure --scan-folder /Users/yifeilin/Library/Developer/Xcode/DerivedData/lgapp-etnuzaqrmdesiefbjjihipkqilio/Build/Products/Release-iphonesimulator/leisure.app/Frameworks --scan-folder /Users/yifeilin/Library/Developer/Xcode/DerivedData/lgapp-etnuzaqrmdesiefbjjihipkqilio/Build/Products/Release-iphonesimulator/leisure.app/PlugIns --scan-folder /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator11.0.sdk/System/Library/Frameworks/CoreLocation.framework --scan-folder /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator11.0.sdk/System/Library/Frameworks/Security.framework --scan-folder /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator11.0.sdk/System/Library/Frameworks/JavaScriptCore.framework --scan-folder /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator11.0.sdk/System/Library/Frameworks/SystemConfiguration.framework --scan-folder /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator11.0.sdk/System/Library/Frameworks/CoreTelephony.framework --scan-folder /Users/yifeilin/project/core/leisure/ios/Frameworks/AMapSearchKit.framework --scan-folder /Users/yifeilin/project/core/leisure/ios/Frameworks/AMapLocationKit.framework --scan-folder /Users/yifeilin/project/core/leisure/ios/Frameworks/AMapFoundationKit.framework --scan-folder /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator11.0.sdk/System/Library/Frameworks/CloudKit.framework --scan-folder /Users/yifeilin/project/core/leisure/ios/Frameworks/MAMapKit.framework --platform iphonesimulator --toolchain /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain --destination /Users/yifeilin/Library/Developer/Xcode/DerivedData/lgapp-etnuzaqrmdesiefbjjihipkqilio/Build/Products/Release-iphonesimulator/leisure.app/Frameworks --strip-bitcode --resource-destination /Users/yifeilin/Library/Developer/Xcode/DerivedData/lgapp-etnuzaqrmdesiefbjjihipkqilio/Build/Products/Release-iphonesimulator/leisure.app --resource-library libswiftRemoteMirror.dylib

wkoutre, 2017-10-04:
@jiazil Just finally switched to XCode 9.

What worked for me:

1. Double-check... Xcode > Preferences > Location Tab > Command Line Tools XCode 9.0
2. `yarn rm react-native-ble-plx`
3. `yarn add react-native-ble-plx@0.6.3`
4. Clean build and build folder
5. Rebuild project and run on device

Cierpliwy, 2017-10-05:
I'm glad it works.

closed 2017-10-05 by Cierpliwy

jiazil, 2017-10-27:
Clean build and build folder??   is this folder ~/Library/Developer/Xcode/DerivedData?or others?

wkoutre, 2017-10-27:
@jiazil Yes, I run `rm -rf ~/Library/Developer/Xcode/DerivedData/*`

You can also try:

`yarn cache clean` and/or `npm cache clean --force`

You should see Carthage rebuilding the ble project successfully after running #3 in my previous comment, as well as `react-native-link react-native-ble-plx`. Once it rebuilds successfully, XCode should link the library properly in the ble post-install script.