dotintent/react-native-ble-plx #143: [iOS] BleClient React/* headers can't be found

jaitaiwan, 2017-09-21:
Running a iOS 10 targeted project with the pod headers added to the search settings and all setup instructions for this package followed. The following error appears on build:

React/Base/RCTBridgeModule.h:12:9: 'React/RCTDefines.h' file not found

kinber95, 2017-09-26:
I had the same issue change:

#import “RCTBridgeModule.h”
#import “RCTEventEmitter.h”
to:

#import <React/RCTBridgeModule.h>
#import <React/RCTEventEmitter.h>
it fix my problem.

jaitaiwan, 2017-09-26:
That’s a completely different legacy react-native issue. Note that the title of this issue already shows “React/“

Cierpliwy, 2017-09-28:
This issue should be resolved with version 0.6.2.

jaitaiwan, 2017-09-28:
I've confirmed the issues still prevail after an `npm install -S react-native-ble-plx` and run of `react-native link` as well as doing a new clean before building in xcode.

zhentan, 2017-10-02:
I'm having the same problem on 0.6.2 as well

Cierpliwy, 2017-10-03:
Can you paste an error? I created clean RN project with version 0.48.3. Did `react-native link react-native-ble-plx` and proceeded from step 7 as mentioned in documentation. Can't reproduce your issue on my machine. Can you check that?

zhentan, 2017-10-03:
I did the following steps on my machine (10.13 + XCode9 + react-native@0.48.4):
1.) `react-native init rnbletest`
2.) `npm install --save react-native-ble-plx`
2.5) fix [this issue](https://github.com/Polidea/RxBluetoothKit/issues/140) before carthage gets to the RxBluetoothKit.xcodeproj file.
3.) `npm install`
4.) `react-native link react-native-ble-plx`

At this point everything command works fine and no errors in output. However I try to do `react-native run-ios` I get an error during link:
`ld: framework not found BleClientManager for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)`

Opening up the BleClientManager project and attempting to build it will result this error:
`In file included from /Users/ztan/Projects/anova/reactnativeApp/rnbletest/node_modules/react-native-ble-plx/ios/BleClient.m:9:
/Users/ztan/Projects/anova/reactnativeApp/rnbletest/node_modules/react-native-ble-plx/ios/BleClient.h:10:9: fatal error: 'React/RCTBridgeModule.h' file not found
#import <React/RCTBridgeModule.h>
        ^~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.`

Let me know if providing full logs/files would be helpful.

Cierpliwy, 2017-10-04:
First of all, thank you for your detailed steps! Without them I coudn't help you.

> I did the following steps on my machine (10.13 + XCode9 + react-native@0.48.4):
1.) react-native init rnbletest
2.) npm install --save react-native-ble-plx
2.5) fix this issue before carthage/compiler gets to the RxBluetoothKit.xcodeproj file.
3.) npm install
4.) react-native link react-native-ble-plx

With version 0.6.3 your 2.5) point shouldn't be necessary.

> Every command up to this point works fine and there are no errors in output. However when I try to do react-native run-ios I get an error during link:
ld: framework not found BleClientManager for architecture x86_64 clang: error: linker command failed with exit code 1 (use -v to see invocation)

You didn't complete whole setup. `react-native link` is not sufficient to use this library. You should continue your setup from point 7 as it is stated in README.md file. 

> Opening up the BleClientManager project and attempting to build it will result this error:
In file included from /Users/ztan/Projects/anova/reactnativeApp/rnbletest/node_modules/react-native-ble-plx/ios/BleClient.m:9: /Users/ztan/Projects/anova/reactnativeApp/rnbletest/node_modules/react-native-ble-plx/ios/BleClient.h:10:9: fatal error: 'React/RCTBridgeModule.h' file not found #import <React/RCTBridgeModule.h> ^~~~~~~~~~~~~~~~~~~~~~~~~ 1 error generated.

You probably opened  `./node_modules/react-native-ble-plx/ios/BleClient.xcodeproj`, not `BleClientManager`. There is no point in doing that. BleClient doesn't build as a standalone static library. It is injected into your RN xcodeproj with `react-native link` command.

zhentan, 2017-10-04:
Thank you for releasing 0.6.3. Everything seems to be working for me now!

Cierpliwy, 2017-10-04:
Please reopen if you have any issues. Closing for now.

closed 2017-10-04 by Cierpliwy

rynop, 2017-10-06:
I'm running into same issue (`'React/RCTBridgeModule.h' file not found #import <React/RCTBridgeModule.h>`) on `v0.6.3` ios v10 target. iphone.

Here are my steps:

```
git clone git@github.com:Polidea/react-native-ble-plx.git
cp -r react-native-ble-plx/examples/ReactBLEScanner .
cd ReactBLEScanner/
yarn add react-native-ble-plx
react-native link react-native-ble-plx
#open xcode, file > open > ~/ReactBLEScanner/ios/Sniffator.xcodeproj
#Do steps 7-11 in ios section of README.md (https://github.com/Polidea/react-native-ble-plx#ios)
react-native run-ios
```

I then get the error. macOS 10.13. xcode Version 9.0 (9A235)

[here is a full log](https://gist.github.com/rynop/74c30434bd58ae958150c04bcb03f158)

Any idea what I'm doing wrong?

zhentan, 2017-10-09:
I had the same problem with the example app. I think the example app is outdated, as last commits were done over a year ago. Creating a new app with `react-native init` instead of copying the example app will work.

jqn, 2018-05-08:
I ran into this issue too, somehow I skipped step 7. Double check the search path in frameworks is correct.