dotintent/react-native-ble-plx #144: XCode 9 Frameworks not compatible - Is there any solution?!

felix-reichel, 2017-10-02:
➜  BleClientManager git:(master) ✗ carthage update
*** Fetching RxSwift
*** Checking out RxBluetoothKit at "3.0.14"
*** Downloading RxSwift.framework binary at "Oxygen.1"
*** Skipped installing RxSwift.framework binary due to the error:
        **"Incompatible Swift version - framework was built with 3.1 (swiftlang-802.0.53 clang-802.0.42) and the local version is 4.0 (swiftlang-900.0.65 clang-900.0.37)."**
*** Checking out RxSwift at "3.6.1"
*** xcodebuild output can be found in /var/folders/r0/n98856xx11jdjyw7s7_ys4x00000gn/T/carthage-xcodebuild.pcfBaj.log
*** Building scheme "RxBlocking-iOS" in Rx.xcworkspace

Cierpliwy, 2017-10-03:
You shouldn't be worried about this warning. It only says that you are using different Swift compiler and carthage should rebuild all your dependencies manually from source code in that case.

closed 2017-10-10 by felix-reichel

NinjaSudo, 2017-10-11:
@Cierpliwy this however, freezes on one of the repos for me leaving me stuck without my bootstrap complete.