dotintent/react-native-ble-plx #145: The following binaries use incompatible versions of Swift

jiazil, 2017-10-03:


Cierpliwy, 2017-10-04:
Please provide detailed description about your error and how to reproduce it.

Abrax20, 2017-10-19:
I have this error when I Build a Debug version of my code
Can  you help me?

> Command

`CopySwiftLibs /Users/username/Library/Developer/Xcode/DerivedData/project-eyfgqjgtlqhurfgzsxxbgailyofn/Build/Products/Debug-iphonesimulator/project.app
    cd /Users/username/Desktop/projects/project/app/ios
    export CODESIGN_ALLOCATE=/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/codesign_allocate
    export DEVELOPER_DIR=/Applications/Xcode.app/Contents/Developer
    export PATH="/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/usr/bin:/Applications/Xcode.app/Contents/Developer/usr/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin"
    export SDKROOT=/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator11.0.sdk
    builtin-swiftStdLibTool --copy --verbose --sign - --scan-executable /Users/username/Library/Developer/Xcode/DerivedData/project-eyfgqjgtlqhurfgzsxxbgailyofn/Build/Products/Debug-iphonesimulator/project.app/project --scan-folder /Users/username/Library/Developer/Xcode/DerivedData/project-eyfgqjgtlqhurfgzsxxbgailyofn/Build/Products/Debug-iphonesimulator/project.app/Frameworks --scan-folder /Users/username/Library/Developer/Xcode/DerivedData/project-eyfgqjgtlqhurfgzsxxbgailyofn/Build/Products/Debug-iphonesimulator/project.app/PlugIns --platform iphonesimulator --toolchain /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain --destination /Users/username/Library/Developer/Xcode/DerivedData/project-eyfgqjgtlqhurfgzsxxbgailyofn/Build/Products/Debug-iphonesimulator/project.app/Frameworks --strip-bitcode --resource-destination /Users/username/Library/Developer/Xcode/DerivedData/project-eyfgqjgtlqhurfgzsxxbgailyofn/Build/Products/Debug-iphonesimulator/project.app --resource-library libswiftRemoteMirror.dylib`

> Error

`error: The following binaries use incompatible versions of Swift:
/Users/username/Library/Developer/Xcode/DerivedData/project-eyfgqjgtlqhurfgzsxxbgailyofn/Build/Products/Debug-iphonesimulator/project.app/Frameworks/BleClientManager.framework/BleClientManager
/Users/username/Library/Developer/Xcode/DerivedData/project-eyfgqjgtlqhurfgzsxxbgailyofn/Build/Products/Debug-iphonesimulator/project.app/project`

radko93, 2017-10-23:
Try cleaning `/Users/username/Library/Developer/Xcode/DerivedData/` folder.

Cierpliwy, 2017-11-30:
You can always open application with XCode and run **Command+Option+K** (clean) and **Command+Option+Shift+K** (clean build folder)

closed 2017-11-30 by Cierpliwy

jay0lu, 2018-04-04:
Try use watchman watch-del-all && rm -rf $TMPDIR/react-* && rm -rf node_modules/ && npm cache clean && npm install to clean the cash. It works for me on xcode9.2.

harshjoshi096, 2019-05-06:
@Cierpliwy I am getting the same issue, and due to this I am unable to run my project in release mode, It is working on debug mode. Can you please help?

Can anyone help? @Abrax20 @jay0lu