dotintent/react-native-ble-plx #147: "Native Module cannot be null"

IainIsCreative, 2017-10-03:
Hey everybody!

I'm relatively new to React Native, so bear with me if there's things I'm not aware of when it comes to this stuff.

Recently I've been wanting to use `react-native-ble-plx` in my React Native project, but when I run the app in the iOS simulator (to see if the app runs), it returns:

`Native module cannot be null`

This is caused when I use `new BleManager` from this package. I'm unsure as to why, but is part of the reason that it's doing this is because I'm using `create-react-native-app`? Or is it something else?

Any ideas? Thanks very much in advance.

Cierpliwy, 2017-10-04:
Please make sure you complete whole setup instructions from README.md file: `react-natve link` + manual steps from point 7.

frankimhof, 2017-10-08:
I had the same issue when I created my app with >create-react-native-app 

and I couldn't find a solution.
Thats why I tried to start with 

>react-native-init

to try with Xcode (not with expo) 
But then I had the issue described here:

https://github.com/Polidea/react-native-ble-plx/issues/149

I hope we can find a solution.

IainIsCreative, 2017-10-09:
I've now closed the issue as I've gone with the `react-native init` route instead, and I've followed the steps to the point it's trying to build, so this isn't so much an issue any more.

`create-react-native-app` is best avoided for this I think.

closed 2017-10-09 by IainIsCreative