dotintent/react-native-ble-plx #149: Carthage build fails, Carthage 0.25.0, Xcode 9.0

frankimhof, 2017-10-08:
Hello

When I install the library automatically/manually (I tried both) and run with Xcode I get the following Error:
>ld: framework not found BleClientManager for architecture x86_64
>clang: error: linker command failed with exit code 1 (use -v to see invocation)

After some research I found that maybe there's a problem with Carthage.
So I created a Cartfile and added GitHub "Polidea/react-native-ble-plx" to it
but when I do
>carthage update

I get the following error:

Building scheme "BleClientManager" in BleClientManager.xcodeproj
Build Failed
	Task failed with exit code 65:
	/usr/bin/xcrun xcodebuild -project /Users/frank/Desktop/playground/ios/Carthage/Checkouts/react-native-ble-plx/ios/BleClientManager/BleClientManager.xcodeproj -scheme BleClientManager -configuration Release -derivedDataPath /Users/frank/Library/Caches/org.carthage.CarthageKit/DerivedData/9.0_9A235/react-native-ble-plx/v0.6.3 -sdk iphoneos ONLY_ACTIVE_ARCH=NO BITCODE_GENERATION_MODE=bitcode CODE_SIGNING_REQUIRED=NO CODE_SIGN_IDENTITY= CARTHAGE=YES build (launched in /Users/frank/Desktop/playground/ios/Carthage/Checkouts/react-native-ble-plx)

This usually indicates that project itself failed to compile. Please check the xcodebuild log for more details: /var/folders/2c/yz87fb5n1jq64d5ypypj371r0000gn/T/carthage-xcodebuild.TQVFxW.log

I also tried 
>carthage update --platform ios 
and
>carthage update --no-use-binaries
but both fail with the same error.

Please help!

frankimhof, 2017-10-09:
closes "149"

frankimhof, 2017-10-09:
Sorry, I made a mistake in point 9 of the "manually install" section of README.md file.
I added the shell command into the wrong place. 
It should look like this:
Run Script
Shell `/bin/sh`
1 `/usr/local/bin/carthage copy-frameworks`

closed 2017-10-09 by frankimhof

LingboTang, 2017-12-12:
I'm getting the same error still