dotintent/react-native-ble-plx #150: warning: directory not found for option '-F/Users/user/development/bleApp/ios/../node_modules/react-native-ble-plx/ios/BleClientManager/Carthage/Build/iOS.' when running in Xcode.

IainIsCreative, 2017-10-09:
I've managed to get further with getting this library to work with iOS, however this one mystifies me. I have followed all the steps necessary (except 11) for getting this to work in iOS. However, I'm getting this error when I run this in Xcode. The directory and files do exist, they've been made, but I don't understand what's going on here. Has anyone got any advice? Thank you in advance!

(NOTE: No, my username on my computer is not 'user!')

frankimhof, 2017-10-09:
Do you have carthage installed?
If not, do this in terminal (you need homebrew to download carthage)
`brew install carthage`
create cart file with terminal in your repository where your .xcodeproj file is
`touch Cartfile`
open the file with a Texteditor and add the line
`GitHub "Polidea/react-native-ble-plx"`
save and then run
`carthage update`

Thats where I get my error 
https://github.com/Polidea/react-native-ble-plx/issues/149
I hope you will succeed

IainIsCreative, 2017-10-16:
UPDATE: I have got it working! Though I will admit, the fault is my own. Also, the clue is in the title of this issue.

That's right, this was **caused by a spelling mistake.** 

Altogether now...

[image]

This issue is now closed. I'm off on a humiliation conga.

closed 2017-10-16 by IainIsCreative

lifenautjoe, 2020-08-28:
What spelling mistake? Can  you please clarify?