dotintent/react-native-ble-plx #151: Read data when application closed

jeandbonicel, 2017-10-10:
hi,

I'm currently working on an Android react native application using this repository to connect by bluetooth some sensors and read data.

I would like to find a way to read the data when my app is closed (to send some push notifications)
but I don't know if it is possible... do you have some examples ? or ideas to do that ?

my code to the connection part:

`import React, { Component } from 'react';
import { 
	Platform,
	View, 
	Text,
	ListView,
	StyleSheet,
	Button,
	TouchableOpacity
   } from 'react-native';

import { BleManager } from 'react-native-ble-plx';

import Container from '../widgets/container';

import CardViewGlobal from '../widgets/cardViewGlobal';
import CardViewSensor from '../widgets/cardViewSensor';
import CustomIcons from '../widgets/customIcons';

import * as Global from '../../global'; 

import * as Constants from '../../constants/sensors';

import { connect } from 'react-redux';

import * as deviceActions from '../../actions/deviceActions';

import CardView from 'react-native-cardview';

const mapStateToProps = (store) => {
  return store;
};

const mapDispatchToProps = (dispatch) => ({
	isConnected: (boolean) => {
        dispatch(deviceActions.isConnected(boolean));
    },
    setDevice: (data) => {
        dispatch(deviceActions.setDevice(data));
    },
    setInfo: (info) => {
        dispatch(deviceActions.setInfo(info));
    },
    setServices: (services) => {
    	dispatch(deviceActions.setServices(services));
    },
    setCharacteristicsForService: (data) => {
    	dispatch(deviceActions.setCharacteristicsForService(data));
    },
    updateCharacteristicValue: (data) => {
    	dispatch(deviceActions.updateCharacteristicValue(data));
    },
    saveHistoricalCharacteristicData: (data) => {
    	dispatch(deviceActions.saveHistoricalCharacteristicData(data));
    },
    setDeviceGlobalValue: (value) => {
    	dispatch(deviceActions.setDeviceGlobalValue(value));
    }
});

class Device extends Component {
[... 349 lines cut ...]
	    )
  	}
}

const styles = StyleSheet.create({
	cardViewSensorList: {

	},
	cardViewGlobal: {

	},
    hidden: {
    	width: 0,
    	height: 0
    },
    minusIcon: {
    	display: 'flex',
    	alignItems: 'flex-end',
    	paddingRight: 15,
    	marginTop: -7,
    	justifyContent: 'space-between'
    }
});

export default connect(mapStateToProps, mapDispatchToProps)(Device);`

dariuszseweryn, 2017-10-11:
Hello,

To run code on Android OS in the background is not a trivial task. There are several things that you need to keep in mind:
1. Android OS may arbitrarily suspend and/or kill your app when it is in background
2. You can implement a Background Service that will run your code (Android <6.0)
3. You may need to implement a Foreground Service that will run your code and show a notification as long as it is running (Android >=6.0)
4. You can utilise a `JobScheduler` (Android >= 5.0) for running a periodic background task

Generally speaking the new Android (8.0+) does not allow any tasks running in the background if they are not a `Foreground Service`, scheduled via `JobScheduler` or your app being whitelisted by the user.

Either way it is unlikely that it will be possible to use the app in the background without writing some native code.

Feel free to describe your case in more detail.

jeandbonicel, 2017-10-11:
OK thanks for you answer! I think you answered to my question :)
I will probably use a foreground service to take care about the bluetooth

closed 2018-01-05 by Cierpliwy