dotintent/react-native-ble-plx #152: monitorCharacteristicForDevice works fine on IOS but not on Android

fogg4444, 2017-10-10:
Attempting to monitor changes and get the listener callback to fire when my hardware notifies the phone.

This code works as expected on IOS but I get no response from the listener on Android.

Followed these docs: [https://polidea.github.io/react-native-ble-plx/#BleManager#monitorCharacteristicForDevice](url)

Is there any known difference in functionality between the platforms? I've been digging around and logging out data in `BleModule.java` to try and get some feedback but I'm not getting anywhere.

Has anyone else had this problem?

These are my versions:
`"react-native": "^0.47.0-rc.1",
    "react-native-ble-plx": "^0.6.3",`

    

The code in question:

    let thisSubscription = this.manager.monitorCharacteristicForDevice(this.connectedDevice.id, this.serviceUUID, this.tagCharacteristicUUID, (err, characteristic) => {
      // console.log('monitorCharacteristicForDevice fires once');
      if(err) {
        console.log('Err from pen', err);
        return
      }
      console.log('data from pen: ', characteristic);
      let tagValue = utils.base64toHEX(characteristic.value)
      console.warn('Tag data: ', tagValue);
    })

    console.log('thisSubscription', thisSubscription);`

elailai94, 2017-11-29:
I'm having this issue as well. I was using the [React Native BLE Manager](https://github.com/innoveit/react-native-ble-manager) library before I switched to this library. The characteristic update listener in that library was also not firing on Android while it works beautifully on iOS. Is there any update on this issue?

Cierpliwy, 2017-11-29:
I definitely need Android's native module log to help you there. Are you sure that you are not missing notifications before you are monitoring them?

wkoutre, 2017-12-05:
This is happening to me, as well. My current workaround for Android is to set the `isReadable` attribute on the characteristic to `true`.

Before reading, `characteristic.value === null`; after calling `const charData = await characteristic.read()` the `value` has the data I expect.

fogg4444, 2017-12-05:
@elailai94 I was using the same library before as well
@Cierpliwy  I'll try and get the native logs up here soon
@wkoutre  I'll give this workaround a try as well. I'm using custom hardware so we'll see if this works out with the software that's embedded

Been bogged with administrative tasks but I'm working on a paid project with this library so I'll be pretty motivated to help fix it sooner or later ;)

Cierpliwy, 2017-12-06:
That would be very helpful @fogg4444 !

mswokk, 2017-12-08:
I guess I solved this issue. 

The problem was RxAndroidBle which is used by `react-native-ble-plx`.
I found it sending BLE notification setup request to BLE device before it is ready to receive it.
I changed order to fix it.
See https://github.com/Polidea/RxAndroidBle/pull/336

To apply this fix, you have to use my own temporal version of `react-native-ble-plx` 
[https://github.com/mswokk/react-native-ble-plx/tree/develop](url)
(Because it's not reviewed yet.)

You can test it follwing 2 steps.
1.  Replace react-native-ble-plx in `package.json`
```
"dependencies": {   
      ...
     "react-native-ble-plx": "github:mswokk/react-native-ble-plx#develop",  // replace this line
      ...
}
```

2. Add jitpack repository in `(your_project)/android/build.gradle`
```gradle
allprojects {
    repositories {
        ....
        maven { url "https://jitpack.io" }  // add this line
```

In my case, I always lost first notification from BLE device just after starting notification.
After this fix, it works well as on iPhone.

If you tried this fix, I hope you to reply here your test result on your device with this fix. 
Thank you

Cierpliwy, 2017-12-08:
Thanks for testing that the issue is caused by RxAndroidBle. I'll talk with @dariuszseweryn how we can proceed with a fix.

Cierpliwy, 2018-01-05:
This should be fixed with version 0.8.0. Please reopen if there are any problems.

OneStromberg, 2018-07-31:
@Cierpliwy 
Hi there! It seems that problem is still there.
I'm using 0.9.1 version of library right now.
Phone Samsung s8, android version 8.0.0.

The goal is to subscribe to several Notification Services. After subscription they start to propagate data. Issue is on iOS everything works correct and on Android I had an issue written here.

I tried to find a core of the problem and found that is I put a break point to first line of **onCharacteristicChanged** at **RxBleGattCallback** - and resume it after stop immitiatly - everything starts to work - I get all messages on my Android Phone.

Thanx!

Cierpliwy, 2018-07-31:
In version 0.8.0 I created a workaround, which caused some issues on our side. That's why I reverted it. I think that this should be fixed in RxAndroidBle directly and probably will look into it on Friday (my open source days).

hoffmannjan, 2018-09-13:
What is recommended solution nowadays? :)

dariuszseweryn, 2018-09-13:
Sit tight, I am currently working on the solution ;)

hoffmannjan, 2018-10-17:
1.0 should resolved this issue? Cause It still occurs in my configuration on Android.

closed 2019-03-23 by stale[bot]

hoffmannjan, 2019-04-08:
@dariuszseweryn In my case the problem is still there, is it really fixed? I'm using __1.0.3__

dariuszseweryn, 2019-04-08:
@hoffmannjan The original poster's issue should be fixed since `1.0.2`. Could you explain your exact use case and add [logs](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) from Logcat?

hoffmannjan, 2019-04-08:
@dariuszseweryn I've created a new issue #446