dotintent/react-native-ble-plx #153: writing back to back characteristic data

scandinaro, 2017-10-11:
My application needs to write three pieces of data to the same service/characteristic.  I'm currently attempting to write them back to back, but only the first one is written successfully.

I first connect to the device successfully with
```javascript
device = await device.connect();
````
then discover
```javascript
device = await device.discoverAllServicesAndCharacteristics();
```
then call a series of 3 functions which each attempts to write with which pass the device variable from one function to the next.
```javascript
await device.writeCharacteristicWithResponseForService('FFFF', 'FF01', data);
```
If I switch the order of the chained functions the first one always gets written correctly while the other two don't seem to make it to the device successfully.   Is there a bug with the library related to writing multiple times in one connection or am I doing something wrong?

Cierpliwy, 2017-10-18:
How do you check that value is written successfully? Are you calling reads between write calls? Are you checking values on peripheral device? Remember that as a result of write call you won't have up to date value stored in Characteristic object on iOS platform.

scandinaro, 2017-10-28:
The issue was completely on the end of my characteristic data and/or device.  My device required a sequence number as part of the write.  The documentation said "always use 00" when in reality the device was expecting sequential sequence numbers if multiple writes were being done within one connection.  Making a change to increment a sequence number and pass that as part of my characteristic write data resolved this issue.

closed 2017-10-28 by scandinaro