dotintent/react-native-ble-plx #154: Error after upgrading react-native to 0.49.3

cltsang, 2017-10-11:
Previously when it's on 0.48.0, everything was fine. Right after upgrading to 0.49.3, when I read state after registering for state change event, i.e. 

```
bleManager.onStateChange((newState) => dispatch(bluetoothStateChanged(newState)));
const currentState = await bleManager.state();
```

I encounter the error

> 2017-10-11 19:15:49.104810+0800 IotApp[699:225767] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x171069a00> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
[RxBLEKit|DEBG|11:15:49.110]: CentralManager(6191225344) didUpdateState(state: poweredOn)
2017-10-11 19:15:49.111 [warn][tid:com.facebook.react.BleClientManagerQueue][RCTEventEmitter.m:54] Sending `StateChangeEvent` with no listeners registered.
2017-10-11 19:15:49.110848+0800 IotApp[699:225284] Sending `StateChangeEvent` with no listeners registered.

Has anyone encountered the same issue? Any advice on where I should begin looking other than downgrading?

Thanks.

closed 2017-10-12 by cltsang

mateusz-, 2017-11-14:
I also ran across this issue. Fixed it by moving instantiation of the `BleManager` to `componentDidMount`.
Before:

```javascript
  constructor() {
    super();
    this.ble = new BleManager();
  }
  componentWillMount() {
    this.ble.onStateChange(status => {
      this.setState({
        status,
      });
    }, false);
  }
  componentDidMount() {
    /* ... */
  }
```
After:
```javascript
  componentDidMount() {
    this.ble = new BleManager();
    this.ble.onStateChange(status => {
      this.setState({
        status,
      });
    }, false);
  }
```

The docs do instantiate BleManager [in the constructor](https://github.com/Polidea/react-native-ble-plx/blob/master/docs/index.html#L916).

lnanhkhoa, 2018-03-26:
Thank you. I solved my problem.

lnanhkhoa, 2018-03-26:
something wrong. i think it need to waiting the `new BleManager()` finish.

rclai, 2018-05-20:
This is still an issue even after putting it in the componentDidMount lifecycle.

In addition, I am also getting warnings such as `Sending `DisconnectionEvent` with no listeners registered` as well.

nandorojo, 2019-11-07:
I fixed this error by going to `ios/projectName/Info.plist` and adding this:

```
	<key>NSBluetoothAlwaysUsageDescription</key>
	<string>Allow app to access your bluetooth</string>
```

I had previously added this directly in my `app.json` => `expo.ios.infoPlist`, but I guess that only works for managed expo projects, and I had to use a bare workflow to use this library.