dotintent/react-native-ble-plx #156: How to write to an unknown service?

lenninlc, 2017-10-17:
So far  I've been able to write and read using the UART service (TX, RX)

Script:

var base64 = require("base-64");

var device;
var manager;

var serviceUUID = "6e400001-b5a3-f393-e0a9-e50e24dcca9e";
var writeUUID = "6e400002-b5a3-f393-e0a9-e50e24dcca9e";
var notifyUUID = "6e400003-b5a3-f393-e0a9-e50e24dcca9e";

export default {
  sendMessage(message) {
    this.device
      .writeCharacteristicWithResponseForService(
        serviceUUID,
        writeUUID,
        base64.encode(message)
      )
      .catch(error => {
        Alert.alert(
          "Device Lost Connection",
          "You lost connection probably. Please bind from main menu to continue."
        );
      });
  },

  listenForMessage(response) {
    this.device.monitorCharacteristicForService(
      serviceUUID,
      notifyUUID,
      (error, res) => {
        if (res != null) response(base64.decode(res.value.toString()));
        else console.log("Disconnected! Please stop!");
      }
    );
  },

  disconnect() {
    if (this.device != null) {
      if (this.device.isConnected()) {
        this.device.cancelConnection();
      }
    }
  },
However, when I tried to do the same with and unknown service It does not work. Do I have to add something on my script for an unknown service? thanks!

Cierpliwy, 2017-10-18:
Did you call [`discoverAllServicesAndCharacteristicsForDevice`](https://polidea.github.io/react-native-ble-plx/#BleManager#discoverAllServicesAndCharacteristicsForDevice)? After that you can check which services/characteristics are available by calling [`servicesForDevice`](https://polidea.github.io/react-native-ble-plx/#BleManager#servicesForDevice)/[`characteristicsForDevice`](https://polidea.github.io/react-native-ble-plx/#BleManager#characteristicsForDevice). Make sure that your unknown service is present there.

closed 2017-11-30 by Cierpliwy

LingboTang, 2017-12-18:
@lenninlc listenForMessage(response) , response is a callback to manipulate the res right?
@Cierpliwy Is monitor a global subscription once turned on?