dotintent/react-native-ble-plx #157: Customizable device name ~

sooxiaotong, 2017-10-18:
Add customizable device name for `startDeviceScan` not always necessary to use device default name.

Cierpliwy, 2017-10-18:
Can you elaborate more? I'm not sure what is your requirement?

sooxiaotong, 2017-10-18:
can we customize our device name with js without changing our phone bluetooth display name?

Cierpliwy, 2017-10-18:
Changing device name must be supported by device itself. You should check if your manufacturer allows that and how to do it. There is common [Device Name](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.gap.device_name.xml) characteristic which can be used to read and in some cases to override device name (if it's writable).

sooxiaotong, 2017-10-18:
oh ya, I got another question, can I get device id ( Mac Addr & UUID ) without opening Bluetooth?

Cierpliwy, 2017-10-19:
Check answer at #92

closed 2017-11-30 by Cierpliwy