dotintent/react-native-ble-plx #159: Undefined is not an object -  new BleManager();

JordanKlaers, 2017-10-18:
I followed the steps from the read.me and am getting an error when calling the constructor function const manager = new BleManager();

the error is  - Undefined is not an object (evaluating  "_BleModule.BleModule.createClient')

I also have the line - `import { BleManager } from 'react-native-ble-plx';` as well

I have tried a few different thing to fix it and the root issue seems to be evading me. Curious what the issue is and how to fix it 

settings.gradle --- 
`include ':app'
include ':react-native-ble-plx'
project(':react-native-ble-plx').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-ble-plx/android')
include ':react-native-ble-plx'
project(':react-native-ble-plx').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-ble-plx/android')
`

build.gradle -- 
`// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
  repositories {
    jcenter()
  }
  dependencies {
    classpath 'com.android.tools.build:gradle:2.1.0'
    compile project(':react-native-ble-plx')
    // NOTE: Do not place your application dependencies here; they belong
    // in the individual module build.gradle files
  }
}

allprojects {
  repositories {
    mavenLocal()
    jcenter()
    maven {
      // Point to local maven repository
      url "$rootDir/../.expo-source/android/maven"
    }
  }
}

task clean(type: Delete) {
  delete rootProject.buildDir
}
`

androidManifest.xml -- 
`<?xml version="1.0" encoding="utf-8"?>
<manifest package="ChromaCrown"
          xmlns:android="http://schemas.android.com/apk/res/android">

          <uses-permission android:name="android.permission.BLUETOOTH"/>
              <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
              <uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION"/>

              
              <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>

              <uses-sdk
        android:minSdkVersion="18"
        />
  <application
    android:name=".MainApplication"
    android:allowBackup="true"
    android:icon="@mipmap/ic_launcher"
    android:label="@string/app_name"
    android:supportsRtl="true">
    <activity android:name=".MainActivity"
              android:configChanges="keyboard|keyboardHidden|orientation|screenSize"
              android:theme="@style/Theme.Exponent.Light"
              android:windowSoftInputMode="adjustResize">
      <intent-filter>
        <action android:name="android.intent.action.MAIN"/>

        <category android:name="android.intent.category.LAUNCHER"/>
      </intent-filter>

      <intent-filter>
        <data android:scheme="exp46862141888c4943bee4fd658bbfe9ea"/>

        <action android:name="android.intent.action.VIEW"/>

        <category android:name="android.intent.category.DEFAULT"/>
        <category android:name="android.intent.category.BROWSABLE"/>
      </intent-filter>
    </activity>
  </application>

</manifest>
`

MainApplication.java --
 `package ChromaCrown;

import android.support.multidex.MultiDexApplication;

import com.facebook.react.ReactPackage;

import java.util.Arrays;
import java.util.List;

// Needed for `react-native link`
// import com.facebook.react.ReactApplication;
import com.polidea.reactnativeble.BlePackage;

public class MainApplication extends MultiDexApplication {

  // Needed for `react-native link`
  public List<ReactPackage> getPackages() {
    return Arrays.<ReactPackage>asList(
        // Add your own packages here!
        // TODO: add cool native modules

        // Needed for `react-native link`
        new MainReactPackage(),
        new BlePackage(),

    );
  }
}
`

radko93, 2017-10-23:
```// NOTE: Do not place your application dependencies here; they belong
// in the individual module build.gradle files```
Wrong gradle file.

JordanKlaers, 2017-10-24:
Thank you, I ended up getting it to work - my issue was that I was using the command "create-react-native-app nameOfApp" and ejecting from that which didnt work. I needed to use "react-native init" and eject from that so I can get an Xcode project file to open with Xcode so that I could properly follow along from step 7.

The instructions could definitely use a bit more but thanfully I got it working!

closed 2017-10-24 by JordanKlaers