dotintent/react-native-ble-plx #161: nothing returned from device.discoverAllServicesAndCharacteristics()

JordanKlaers, 2017-10-24:
I have a raspBerry Pi zero W running node with bleno. 
I have a service for writing to the PI. 
I used lightblue to connect to the pi over bluetooth and write hex values which I am able to see logging from the pi. (<< just to confirm the pi is working as expected)

So I am sure I have a service working but when I call device.discoverAllServicesAndCharacteristics()
from the "Connecting and discovering services and characteristics" section in the documents, the device object, when I console.log from the second .then()  looks like --
id:"B8:27:EB:E4:EF:C5"
isConnectable:null
localName:null
manufacturerData:null
name:"raspberrypi"
overflowServiceUUIDs:null
rssi:null
serviceData:null
serviceUUIDs:null
solicitedServiceUUIDs:null
txPowerLevel:null

to me it seems there are no services coming from the device? what might I be doing incorrectly?
I have also confirmed the deviced id matches that coming from the pi.

Cierpliwy, 2017-10-24:
There is `getServices()` function which returns Array of currently discovered services.

JordanKlaers, 2017-10-25:
Thank you, I was able to get the services and eventually characteristics and I can write to the device. I noticed you cant send large amount of data in a single call. When i try to send the string "Hello world how long can this be?" the other end only received "hello world how long". I am curious if there is someway to send more data or do I need to make multiple calls if I want more data sent?

Cierpliwy, 2017-10-25:
That's basically it. Both devices negotiate MTU which is used for further transmission. Minimum value is 23, so with 3 bytes ATT header you can send 20 bytes. iOS by default negotiates MTU of size 158 and on Android you can call `device.requestMTU(your_value)` to try get bigger payload per package.

pstanton, 2017-11-22:
i can't find `device.getServices()` but `device.services()` .. i assume that's what you mean?

Cierpliwy, 2017-11-30:
Fixed previous comment to not mislead anybody.

closed 2017-11-30 by Cierpliwy