dotintent/react-native-ble-plx #164: Could not figure out the format of the value returned after connecting and reading a specific  characteristic

KarthickCSK, 2017-10-27:


Cierpliwy, 2017-10-27:
You should check documentation before asking questions:
https://polidea.github.io/react-native-ble-plx/#characteristic
Values are in Base64 format.

KarthickCSK, 2017-10-27:
Yeah . Apologies  but i am unable to convert the Value i mentioned above from base64 to integer. Gets an NaN error using base64-int32 module

Cierpliwy, 2017-10-28:
Your value looks fine and can be decoded as LE or BE int32 value:
```
var base64 = require("base64-js")
const array = base64.toByteArray("CAIBAQAAAAAAAAAAAAAAAAAAAAA=").buffer
const intLE = new DataView(array).getInt32(0, true)
const intBE = new DataView(array).getInt32(0, false)
console.log("LE: ", intLE, ", BE: ", intBE)
```

closed 2017-11-30 by Cierpliwy